google/skywater-pdk #1: Publish OpenAccess files to improve Cadence Virtuoso support

mithro, 2020-05-07:
The first publication of this is missing files required for optimal Cadence Virtuoso usage. These files will be released publically at a future date after further work.

These files are;
 * OpenAccess versions of the cells
 * Cadence PCells, including
     - Parameterized primitives including FETs, Capacitors, Resistors, Inductors, etc
     - Seal Ring
 * Cadence SKILL scripts that support Virtuoso functions such as netlisting

Before these files are released access to these files is available through SkyWater directly under NDA.

very3b, 2020-06-19:
willing to contribute~ @mithro

mithro, 2020-06-19:
@very3b please email me at tansell@google.com and we can discuss options for how you can help. @msaligane and @stineje are also looking at things in this area.

msaligane, 2020-06-19:
@very3b I saw your request to get access to the private/cad repo of FASoC. There are gf12lp pdk info + other tool confidential info we can't really share. Can you please share your affiliation so I could check?

patrickschulz, 2020-07-07:
I would also like to contribute to the analog part of the PDK and general integration of pcells/cell generation independent of cadence virtuoso to enable a truly open source PDK.

promach, 2020-11-22:
Does skywater130 have something like https://www.eda.ncsu.edu/wiki/FreePDK15:Analog_Artist_with_HSPICE#Create_Aliases_to_Setup_Your_Environment to setup library for cadence virtuoso ?

b224hisl, 2023-08-07:
so，， can Cadence Virtuoso load  the sky130A PDK successfully now? I still met the problem when loading the tlef
[image]

b224hisl, 2023-08-09:
> @b224hisl : The ".tlef" file is a technology LEF file used with digital synthesis. It is not a technology file for Cadence. Because Cadence formats are Cadence-proprietary, they could not be open-sourced and so are not part of the open-source PDK.

Sorry I'm new in this filed and I don't get it. Actually I'm creating sky130A's library in virtoso, which is asking for technology file. How can I load sky130A's cell into virtoso?

RTimothyEdwards, 2023-08-09:
The sky130 PDK doesn't support Cadence.  If you want to use Cadence, you should sign an NDA with SkyWater directly and get the S130 proprietary PDK.

b224hisl, 2023-08-10:
> The sky130 PDK doesn't support Cadence. If you want to use Cadence, you should sign an NDA with SkyWater directly and get the S130 proprietary PDK.

I think S130 is totally opensource. Because I used to use the technology file in innovus for PnR in chip designing successfully. Do you mean S130 has another version?

toufiqhanik, 2023-10-18:
Hello @RTimothyEdwards,

I am trying to generate a macro design with manual placement and routing using Skywater130 (sky130_fs_sc_hd) in Cadence Innovus to validate our design concept. I found that after the automatic early global route/ nano route, some of the cells i/o are not connected properly (this step is okay for some other libraries). Also, there are some DRC violations but not showing properly in the violation browser in GUI mode. Moreover, found this library not very friendly/optimized to do manual P&R. Even automatic P&R is not optimized and working properly in Cadence Innovus. Maybe I am making some mistakes in my flow. However, do you think these available versions are still not supported in Innovus?

Moreover, I see that the full library is available under NDA. Can you please guide me on how can I get the full version so that I can use it in Innovus for some academic research?

RTimothyEdwards, 2023-10-18:
@toufiqhanik :  Sounds like you want the s130 PDK directly from SkyWater.

VardhanSuroshi, 2023-12-09:
@RTimothyEdwards  are we now able to use the SKYwater 130 PDK within cadence? 
@toufiqhanik  @b224hisl Were you successful in integrating and using SKY130?  If so could please share how to integrate it

RandyXi, 2025-08-07:
> > The sky130 PDK doesn't support Cadence. If you want to use Cadence, you should sign an NDA with SkyWater directly and get the S130 proprietary PDK.
> 
> I think S130 is totally opensource. Because I used to use the technology files in innovus for PnR in chip designing successfully. Do you mean S130 has another version?

I am facing the same issues with Cadence virtuoso to work with S130, what I came across is :
adence Virtuoso Support[¶](https://skywater-pdk.readthedocs.io/en/main/known_issues.html#cadence-virtuoso-support)
The first publication of this is missing files required for optimal Cadence Virtuoso usage. These files will be released publicly at a future date after further work (see [Issue #1](https://github.com/google/skywater-pdk/issues/1)).

"https://skywater-pdk.readthedocs.io/en/main/known_issues.html#cadence-virtuoso-support"

you can have a review, if you come across any PDK that can do production and or prototyping that doesn't requires an NDA, please let me know, i am also knew to this and i am using "**https://eda.ncsu.edu/design-kits/** this PDK is usable but not certain if productive or can do prototyping, hope you come across a resolve...

RandyXi, 2025-08-08:
> [@RandyXi](https://github.com/RandyXi) : As of now (which invalidates the comments from a couple of years ago), Cadence has its own version of the open PDK which is based on sky130 and not S130. We just did a test shuttle run with SkyWater which should be heading to SkyWater now or in the next few days. That would be the preferred PDK to use if you have Cadence tools and want to use the open PDK.

@RTimothyEdwards much appreciation for your prompt and insightful info, I didn't know they wss a sky130 unless i overlook it for S130, I'm designing in Cadence OrCAD/Allegro and i stepping into new waters in IC design and its pretty interesting, if any advice or weblink to guide me into bigger waves please info me, thanks once again

RTimothyEdwards, 2025-08-08:
@RandyXi :  The only other advice I have is to join the open source silicon community at https://fossi-chat.org.

buttercutter, 2025-11-02:
Could we use https://github.com/stineje/sky130_cds for analog and RF IC design in Virtuoso ?  Are there any limitations for this specific github version ?