google/skywater-pdk #10: Publish the SKY130 Oklahoma State University provided standard cell library

mithro, 2020-05-07:
The SKY130 Oklahoma State University provided standard cells library is currently empty, only a placeholder is currently provided.

The SKY130 Oklahoma State University provided standard cells library will be integrated at a future date.

Before these files are released access to these files is available at https://github.com/stineje/OSU_130_PDK

stineje, 2020-06-18:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe
>
> https://github.com/stineje/OSU_130_PDK<https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fstineje%2FOSU_130_PDK&data=02%7C01%7Cjames.stine%40okstate.edu%7C9d6998d197e548b3572e08d813a0428f%7C2a69c91de8494e34a230cdf8b27e1964%7C0%7C0%7C637280926317611187&sdata=TE8KpXyaqIWAJMSHC4IzqmQgSqMHggu8xcT%2FPHHke3Y%3D&reserved=0> is not working.

Niels,

Thanks for the Email.  OSU_130_PDK is a private repo with only a set number of approved users.  Let me know your relationship with us and we will do what we can.  Sorry for the issues.  Take care.

All my best,

James

-----------------------------------------------------
James E. Stine, Jr., Ph.D.
Earl and Carolyn Glimp Professor in Engineering
Oklahoma State University
Department of Electrical and Computer Engineering
VLSI Computer Architecture Research Group
215 General Academic Building
Stillwater, OK 74078 USA
E-mail: james.stine@okstate.edu<mailto:james.stine@okstate.edu>
Phone: (405) 744-9244
Fax: (405) 744-9198
Research URL: http://vlsiarch.ecen.okstate.edu
BouldinList URL: http://bouldinlist.ecen.okstate.edu<http://bouldinlist.ecen.okstate.edu/>
-----------------------------------------------------

RTimothyEdwards, 2020-06-18:
@trcwm :  But, it will be made public soon.  Everything is still in motion right now as we prepare files for the distribution.  Public announcements about the skywater_pdk repo will start in a week or so, and then we will open up repositories to the public as soon as we are able to.

trcwm, 2020-06-18:
@RTimothyEdwards Thanks Tim, I'm looking forward to testing the first open-source PDK in the world!