google/skywater-pdk #3: Publish the SKY130 SRAM Build Space

mithro, 2020-05-07:
The SRAM build space is currently empty, only a placeholder is currently provided.

The components of the SRAM build space (compatible with OpenRAM) will be released at a future date.

Before these files are released access to these files is available through SkyWater directly under NDA.

zhongweidee, 2021-03-27:
it must have NDA with skyWater ? it's not publish for even trial?

mithro, 2021-03-27:
This **_should_** have already been released but there have been some delays with the memory compiler.

Ravenslofty, 2022-02-27:
Since it's been roughly a year, could I have an update on this?

mithro, 2022-02-27:
This could be better documented but the build space has been published.

The SRAM build space itself was published at https://github.com/google/skywater-pdk-libs-sky130_fd_bd_sram back in October last year.

The configuration to use this build space with OpenRAM memory compiler is included directly in the OpenRAM repository at https://github.com/VLSIDA/OpenRAM/tree/stable/technology/sky130

Some precompiled SRAM macros are published by efabless at https://github.com/efabless/sky130_sram_macros

jalcim, 2023-08-06:
Its nice, but OpenRAM is very bad documented.