google/skywater-pdk #8: Publish script which recombined the individual cell liberty files into a single liberty library

mithro, 2020-05-07:
The skywater-pdk publishes files split into individual cells and models. This makes it easier to see what is available but produces a large number of files and directories and these large number of files makes a number of tools unhappy.

The skywater-pdk repository has Python scripts to combined cells and models inside a library into a single file to make it easier to use with these tools.

This script will be published at a later date.

mithro, 2020-05-07:
This should include documentation on how to run and use these scripts.

thesourcerer8, 2020-06-17:
libertymerge is already capable of doing that: https://codeberg.org/tok/librecell/src/branch/master/librecell-lib should I provide documentation for it?

closed 2020-07-06 by mithro