google/skywater-pdk #11: Publish examples of using the PDK for digital design

mithro, 2020-05-08:
The SkyWater PDK documentation should include examples of using the PDK for digital design with popular tools.

hash-ashin, 2021-12-10:
I am so desperate to see how digital design would work with the PDK... I am just overwhelmed with excitement since I discovered this project this week and I am inevitably going to open up some of the software. If I come up with anything even vaguely presentable, I'll have to publish it because there is a severe deficit here right now!

QuantamHD, 2021-12-11:
Please do we need all the help we can get, but to get you started https://github.com/The-OpenROAD-Project/OpenLane is an RTL -> GDSII flow that we support that target this PDK.

hash-ashin, 2021-12-11:
> Please do we need all the help we can get, but to get you started https://github.com/The-OpenROAD-Project/OpenLane is an RTL -> GDSII flow that we support that target this PDK.

Yes, I've been obsessively watching videos related to this PDK on YouTube and I realized this. I'm not a hardware expert by any means but as long as I can rely on digital abstractions, I might be able to come up with something. Making my own ICs has been a fantasy for several years now so this project is like a dream come true. I just have to answer the last exam of my undergrad on Tuesday then this project is my first priority.