google/skywater-pdk #12: Publish documentation for using the SkyWater PDK for analog design

mithro, 2020-05-08:
The SkyWater PDK documentation should include examples of using the PDK for analog design with popular tools.

lhrodovalho, 2020-07-05:
Is the PDK expected to work with Cadence tools

mithro, 2020-07-05:
@lhrodovalho See also issue #1 

While the open tools are still behind the proprietary tools like Cadence Virtuoso, anyone can contribute to support for them so they have been the first targets.

We do hope to have full support for the Cadence tooling in the future. However due to limited access to these tools there is only a small number of people who can work on this and hence work on them is proceeding at a slower pace.

verntigo, 2020-07-12:
So, what FOSS tools are being targeted for analog design? I have some suggestions.

For schematics:
- GEDA's gschem
- XCircuit
- Electric

For layout (first 2 are obvious):
- Magic
- KLayout
- Xic

For simulation:
- ngspice
- Berkeley SPICE3
- Xyce
- Qucs

For physical verification, DRC and LVS will probably be tied to the layout tool. Parasitic extraction is a standing problem for FOSS tools.

gvidasdj, 2020-09-25:
So who is writing the documentation how to make analog design? Any progress?

aikudo, 2021-03-07:
Is this project dead in the water?

umarcor, 2021-03-12:
@pepijndevos has been looking into Xyce recently (https://github.com/NyanCAD/XyceSimServer). However, I think that simulation with this PDK is not ready to use yet: https://twitter.com/pepijndevos/status/1367713525758648320.

pepijndevos, 2021-03-12:
The xyce people have told me sky130 is high on their priority list, but their tool has better compatibility with hspice/spectre than ngspice.

I've been thinking about @mithro s comment regarding cadence. I assume the proprietary sky130 pdk was written for commercial tools, so would it not be a matter of releasing the original files?

engrvip, 2021-05-26:
Can anyone suggest how to use skywater130 pdk with open source tools for strictly analog design. Is there any reference or video available which can guide on analog IC design using open-source tools with sky130 pdk.

pepijndevos, 2021-05-26:
It can be a bit rough to get started in my experience.
What is your background?
It seems the current tools for open source analog design are quite old and have quite a barrier to entry.
Most people seem to be doing analog design with xschem and magic with a collection of scripts for simulation.
There are some tutorials out there for both magic and xschem.
For installing the PDK open_pdks is the recommended approach.
For getting them on the shuttle, you have to basically just edit the GDS of the provided project.

slagernate, 2023-01-17:
@pepijndevos do you have links you can share regarding the scripts for simulation?

pepijndevos, 2023-01-17:
Well this is what I've been playing with mostly https://github.com/NyanCAD/Pyttoresque-templates