google/skywater-pdk #21: Error in Stackup?

thesourcerer8, 2020-06-17:
In the stackup diagram:
https://raw.githubusercontent.com/google/skywater-pdk/0bd4367203243e40d46a8835670db950c1f339a5/docs/_static/metal_stack.svg
I see 2 different vias called "via2" and "via3", is that a copy&paste error?

RTimothyEdwards, 2020-06-17:
@thesourcerer8 :  Yes, that's an error, thanks for pointing it out.  It comes from the fact that the MiM caps were added to the current process from an earlier process variant that had a different metal stack, so it was not obvious which metal layers the MiM caps are located above.  I moved them up one metal layer and didn't change the via names.

RTimothyEdwards, 2020-06-17:
@mithro :  I have a corrected SVG file;  does it go in this repo, or is it originating from elsewhere?

mithro, 2020-06-17:
@RTimothyEdwards You'll need to send pull requests.

mithro, 2020-06-17:
FYI -- The SVG hasn't been merged yet, it is in the pull request at https://github.com/google/skywater-pdk/pull/19

RTimothyEdwards, 2020-06-17:
@mithro :  I suppose I must wait for you to merge the pull request before I can then update the drawing and make a new pull request?

mithro, 2020-06-17:
The fixed metal_stack.svg file from @RTimothyEdwards is now in the pull request #19

closed 2020-06-17 by mithro