google/skywater-pdk #37: lib Errors when using closed tools

msaligane, 2020-07-02:
This is a list of errors reported by pt: 
  - The 'delay_model' should be the 1st attribute in the library
  -  The 'technology' should be the 1st attribute in the library
-> expectation for this should be something like this:
```
library ("lib_name") {
	define(switching_power_split_model,library,string);
	define(driver_model,library,string);
	define(leakage_sim_opt,library,string);
	define(default_constraint_arc_mode,library,string);
	define(min_pulse_width_mode,library,string);
	define(default_arc_mode,library,string);
	define(def_sim_opt,library,string);
	define(simulator,library,string);
	define(violation_delay_degrade_pct,timing,string);
	define(sim_opt,timing,string);
	**technology("cmos");**
	**delay_model : "table_lookup";**
```

  - The 'lu_table_template' construct is not valid in 'generic_cmos' libraries.
-> I am not sure about this but after verifying versus other libs. The variable comes before the index:
```
 lu_table_template ("constraint_3_0_1") {
        variable_1 : "related_pin_transition";
        index_1("1, 2, 3");
```
       
  - The 'normalized_driver_waveform' construct is not valid in 'generic_cmos' libraries. 
  - The 'cell_fall' construct is not valid in 'generic_cmos' libraries. 

These errors seems to be the source of multiple other errors. So I think they should be fixed first.

mithro, 2020-07-02:
@msaligane -- You can't have two 1st :-P

Do you have a "canonical ordering" that the liberty file attributes should be in?

kkiningh, 2020-07-02:
Synopsys has an example template for logic libraries that might be useful.

```
library (name_string) { 
  /* Library-Level Simple and Complex Attributes */
  technology (name_enum ) ;
  delay_model : "model" ;
  bus_naming_style : "string" ;
  date : "date" ;
  comment : "string" ;
  time_unit  : "unit" ;
  voltage_unit : "unit" ; 
  leakage_power_unit : "unit" ;
  current_unit : "unit" ;
  pulling_resistance_unit : "unit" ;
  capacitive_load_unit (value, unit) ;
  define_cell_area (area_name, resource_type) ;
  revision : float | string ;
  /* Default Attributes and Values (not shown here) */
  /* Scaling Factors Attributes and Values (not shown here) */
  /* Library-Level Group Statements */
  operating_conditions (name_string)  {   ... operating conditions description ... }
  wire_load (name_string)  {    ... wire load description ... }
  wire_load_selection (name_string)  {   ... wire load selection criteria... }
  power_lut_template (namestring)  {   ... power lookup table template information... }
  /* Cell definitions */ 
  cell (namestring2)  {   ... cell description ... }
  ...
  type (namestring) {   ... type description ... }
  input_voltage (name_string) {   ... input voltage information ... }
  output_voltage (name_string)  {   ... output voltage information ... }
}
```

msaligane, 2020-07-02:
This is an example of what I would expect to see in the liberty file:
```
library ("sky130_fd_sc_hd__ff_100C_1v65") {
	define(switching_power_split_model,library,string);
	define(driver_model,library,string);
	define(leakage_sim_opt,library,string);
	define(default_constraint_arc_mode,library,string);
	define(min_pulse_width_mode,library,string);
	define(default_arc_mode,library,string);
	define(def_sim_opt,library,string);
	define(simulator,library,string);
	define(violation_delay_degrade_pct,timing,string);
	define(sim_opt,timing,string);
	technology("cmos");
	delay_model : "table_lookup";
	bus_naming_style : "%s[%d]";
	in_place_swap_mode : "match_footprint";
	library_features("report_delay_calculation");
	switching_power_split_model : "true";
	simulation : true;
	revision : "1.0000000";
	driver_model : "ramp";
	leakage_sim_opt : "runlvl=5 accurate=1 method=bdf kcltest=1 gmin=1E-15 runlvl=5 accurate=1 method=bdf kcltest=1 gmin=1E-15";
	input_threshold_pct_rise : 50.000000;
	input_threshold_pct_fall : 50.000000;
	output_threshold_pct_rise : 50.000000;
	output_threshold_pct_fall : 50.000000;
	slew_lower_threshold_pct_fall : 20.000000;
	slew_lower_threshold_pct_rise : 20.000000;
	slew_upper_threshold_pct_fall : 80.000000;
	slew_upper_threshold_pct_rise : 80.000000;
	slew_derate_from_library : 1.000000;
	time_unit : "1ns";
	voltage_unit : "1V";
	current_unit : "1mA";
	leakage_power_unit : "1nW";
	pulling_resistance_unit : "1kohm";
	capacitive_load_unit(1.000000, \
	  "pf");
	default_fanout_load : 0.000000;
	default_inout_pin_cap : 0.000000;
	default_input_pin_cap : 0.000000;
	default_output_pin_cap : 0.000000;
	default_max_transition : 1.500000;
	default_leakage_power_density : 0.000000;
	default_cell_leakage_power : 0.000000;
	default_constraint_arc_mode : "worst";
	min_pulse_width_mode : "max";
	default_arc_mode : "worst_edges";
	nom_process : 1.000000;
	nom_temperature : 25.000000;
	nom_voltage : 1.800000;
	def_sim_opt : "POST=1 PROBE POST_VERSION=2001 INGOLD=2 NOMOD NOPAGE NUMDGT=10 MEASDGT=10 LIMPTS=500000 ICSWEEP=0 AUTOSTOP ALTCC=1 RUNLVL=5 ACCURATE=1 ABSVAR=1.000000e-03 RELVAR=1.000000e-02";
	simulator : "HSPICE -- F-2011.09-SP2 32-BIT (Feb 27 2012)";
	voltage_map("VSS",0.000000);
	voltage_map("kapwr",1.800000);
	voltage_map("lowlvpwr",1.800000);
	voltage_map("vgnd",0.000000);
	voltage_map("vnb",0.000000);
	voltage_map("vpb",1.800000);
	voltage_map("vpwr",1.800000);
	voltage_map("vpwrin",1.800000);
	operating_conditions ("ff_100C_1v65") {
		process : 1.000000;
		temperature : 25.000000;
		voltage : 1.800000;
		tree_type : "balanced_tree";
	}
```

For the tables constructs:
```

	lu_table_template ("del_1_7_7") {
		variable_1 : "input_net_transition";
		index_1("1, 2, 3, 4, 5, 6, 7");
		variable_2 : "total_output_net_capacitance";
		index_2("1, 2, 3, 4, 5, 6, 7");
	}
	power_lut_template ("power_outputs_1") {
		variable_1 : "input_transition_time";
		index_1("1, 2, 3, 4, 5, 6, 7");
		variable_2 : "total_output_net_capacitance";
		index_2("1, 2, 3, 4, 5, 6, 7");
	}
	lu_table_template ("constraint_3_0_1") {
		variable_1 : "related_pin_transition";
		index_1("1, 2, 3");
	}
	lu_table_template ("vio_3_3_1") {
		variable_1 : "related_pin_transition";
		index_1("1, 2, 3");
		variable_2 : "constrained_pin_transition";
		index_2("1, 2, 3");
	}
	power_lut_template ("power_inputs_1") {
		variable_1 : "input_transition_time";
		index_1("1, 2, 3, 4, 5, 6, 7");
	}
	lu_table_template ("driver_waveform_template") {
		variable_1 : "input_net_transition";
		index_1("1, 2");
		variable_2 : "normalized_voltage";
		index_2("1, 2");
	}
	normalized_driver_waveform ("driver_waveform_template") {
		driver_waveform_name : "ramp";
		index_1("0.0100000, 0.0230506, 0.0531329, 0.1224745, 0.2823108, 0.5000000, 0.6507428, 1.5000000");
		index_2("0.0000000, 0.5000000, 1.0000000");
		values("0.0000000e+00, 8.3333333e-03, 1.6666667e-02", \
		  "0.0000000e+00, 1.9208818e-02, 3.8417635e-02", \
		  "0.0000000e+00, 4.4277440e-02, 8.8554881e-02", \
		  "0.0000000e+00, 1.0206207e-01, 2.0412415e-01", \
		  "0.0000000e+00, 2.3525901e-01, 4.7051801e-01", \
		  "0.0000000e+00, 4.1666667e-01, 8.3333333e-01", \
		  "0.0000000e+00, 5.4228568e-01, 1.0845714e+00", \
		  "0.0000000e+00, 1.2500000e+00, 2.5000000e+00");
	}
```

After fixing normally we will have all the errors gone.

kkiningh, 2020-07-02:
For future reference, the Synopsys Logic Library Reference Manual gives an overview of how Synopsys tools expect things to be arranged in the liberty file. Unfortunately, I can't share it publicly but there's probably an old version floating around somewhere.

mithro, 2020-07-02:
@msaligane I'm going to split this into a couple of seperate issues.

mithro, 2020-07-02:
FYI - The order of the attribute output is defined at https://github.com/google/skywater-pdk/pull/34/files#diff-f9cb759bb20be5053e4a63e7e0b282d2R548-R563 and https://github.com/google/skywater-pdk/pull/34/files#diff-f9cb759bb20be5053e4a63e7e0b282d2R355-R372

msaligane, 2020-07-03:
Still seeing some of the issues reported yesterday. This is out of the log file when reading into primetime:
```
Error: Line 25, The 'delay_model' should be the 1st attribute in the library. (LBDB-292)
Error: Line 33, The 'lu_table_template' construct is not valid
	in 'generic_cmos' libraries. (LBDB-126)
Error: Line 34, The 'index_1' complex attribute cannot be specified here. (LBDB-76)
Error: Line 35, The 'variable_1' attribute cannot be specified here. (LBDB-76)
```

When I checked the .lib, I do not see changes to the order of attributes. I did not get a chance to look at the code unfortunately.

mithro, 2020-07-03:
@msaligane - I haven't actually fixed them yet, I was just pointing out where in the code it should be fixed.

msaligane, 2020-07-03:
@mithro I may give it a try late afternoon to fix it.

mithro, 2020-07-03:
Keep in touch via slack -- I might beat you to it.

mithro, 2020-07-03:
@msaligane -- Do you know what goes into the following sections?
```
  /* Default Attributes and Values (not shown here) */
  /* Scaling Factors Attributes and Values (not shown here) */
```

msaligane, 2020-07-03:
Something like this:
```
 /* default attributes */
 default_cell_leakage_power: 0.0;
 default_fanout_load: 1.0;
 default_output_pin_cap: 0.0;
 default_inout_pin_cap: 0.1;
 default_input_pin_cap: 0.1;
 default_max_transition: 1.0;
```

Scaling factors:
```
k_process_cell_rise
• k_temp_cell_rise
• k_volt_cell_rise
• k_process_cell_fall
• k_temp_cell_fall
• k_volt_cell_fall
• k_process_rise_propagation
• k_temp_rise_propagation
• k_volt_rise_propagation
• k_process_fall_propagation
• k_temp_fall_propagation
• k_volt_fall_propagation
• k_process_rise_transition
• k_temp_rise_transition
• k_volt_rise_transition
• k_process_fall_transition
• k_temp_fall_transition
• k_volt_fall_transition
```

msaligane, 2020-07-03:
@mithro My fix works should I push it or not?

mithro, 2020-07-04:
How does the following order look?
```
()                                         34    0 --  define                                   ''                                       [{'attribute_name':  ...
()                                         61    0 --  comp_attribute                           'technology'                             cmos ...
()                                         90    0 --  delay_model                              ''                                       table_lookup ...
()                                        119    0 --  bus_naming_style                         ''                                       %s[%d] ...
()                                        199    0 --  time_unit                                ''                                       1ns ...
()                                        225    0 --  voltage_unit                             ''                                       1V ...
()                                        253    0 --  leakage_power_unit                       ''                                       1nW ...
()                                        287    0 --  current_unit                             ''                                       1mA ...
()                                        315    0 --  pulling_resistance_unit                  ''                                       1kohm ...
()                                        354    0 --  comp_attribute                           'capacitive_load_unit'                   [1.0, 'pf'] ...
()                                        446    0 --  revision                                 ''                                       1.0 ...
()                                        484    0 --  default_cell_leakage_power               ''                                       0.0 ...
()                                        525    0 --  default_fanout_load                      ''                                       0.0 ...
()                                        559    0 --  default_inout_pin_cap                    ''                                       0.0 ...
()                                        595    0 --  default_input_pin_cap                    ''                                       0.0 ...
()                                        631    0 --  default_max_transition                   ''                                       1.0 ...
()                                        668    0 --  default_output_pin_cap                   ''                                       0.0 ...
()                                        705    0 --  default_arc_mode                         ''                                       best_edges ...
()                                        705    0 --  default_constraint_arc_mode              ''                                       worst ...
()                                        705    0 --  default_leakage_power_density            ''                                       0.0 ...
()                                       1341    0 --  operating_conditions                     'ff_100C_1v65'                           {'process': 1.0, 'te ...
()                                       1601    0 --  power_lut_template                       'hidden_pwr_template13'                  {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x7'                       {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x10'                      {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x11'                      {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x12'                      {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x13'                      {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x14'                      {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x15'                      {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x16'                      {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x17'                      {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x18'                      {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x19'                      {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x20'                      {'index_1': [1, 2, 3 ...
()                                       1601    0 --  power_lut_template                       'pwr_template13x21'                      {'index_1': [1, 2, 3 ...
[... cut ...]
()                                        inf  inf --  slew_upper_threshold_pct_fall            ''                                       80.0 ...
()                                        inf  inf --  slew_upper_threshold_pct_rise            ''                                       80.0 ...
()                                        inf  inf --  switching_power_split_model              ''                                       true ...
()                                        inf  inf --  tran_timestep                            ''                                       1e-12 ...
```

kkiningh, 2020-07-04:
To answer the question in the [FIXME](https://github.com/mithro/skywater-pdk/commit/2ac296e3762925d7701dc04caffda265022aaf13#diff-f9cb759bb20be5053e4a63e7e0b282d2R446), the `type` group is for defining structural constraints for buses. The `input_voltage` and `output_voltage` groups are for describing either the voltage levels a pad is driven at or the voltages ranges for standard cells. I don't think any of these are required though.

Examples: 
```
type ( BUS4 ) {
  base_type : array ;
  data_type : bit ;
  bit_width : 4 ;
  bit_from : 0 ;
  bit_to : 3 ;
  downto :false ;
}
```

```
library (my_library) {
...
   input_voltage(CMOS) {
     vil : 0.3 * VDD;
     vih : 0.7 * VDD;
     vimin : -0.5;
     vimax : VDD + 0.5;
   }
}
```

msaligane, 2020-07-04:
@mithro this seems to be fixed in this revision.

commit ee775979 (2020-07-06): api: Change the way liberty attributes are sorted.

Fixes #37, #40.

The `LIBERTY_ATTRIBUTE_ORDER` template is used to control the ordering
of the liberty attribute output.
closed 2020-07-04 by msaligane