google/skywater-pdk #40: Table constructs in the generated liberty files cause warnings

mithro, 2020-07-02:
Originally reported in https://github.com/google/skywater-pdk/issues/37

The table constructs should have a form which looks like this;
```
	lu_table_template ("del_1_7_7") {
		variable_1 : "input_net_transition";
		index_1("1, 2, 3, 4, 5, 6, 7");
		variable_2 : "total_output_net_capacitance";
		index_2("1, 2, 3, 4, 5, 6, 7");
	}
	power_lut_template ("power_outputs_1") {
		variable_1 : "input_transition_time";
		index_1("1, 2, 3, 4, 5, 6, 7");
		variable_2 : "total_output_net_capacitance";
		index_2("1, 2, 3, 4, 5, 6, 7");
	}
	lu_table_template ("constraint_3_0_1") {
		variable_1 : "related_pin_transition";
		index_1("1, 2, 3");
	}
	lu_table_template ("vio_3_3_1") {
		variable_1 : "related_pin_transition";
		index_1("1, 2, 3");
		variable_2 : "constrained_pin_transition";
		index_2("1, 2, 3");
	}
	power_lut_template ("power_inputs_1") {
		variable_1 : "input_transition_time";
		index_1("1, 2, 3, 4, 5, 6, 7");
	}
	lu_table_template ("driver_waveform_template") {
		variable_1 : "input_net_transition";
		index_1("1, 2");
		variable_2 : "normalized_voltage";
		index_2("1, 2");
	}
	normalized_driver_waveform ("driver_waveform_template") {
		driver_waveform_name : "ramp";
		index_1("0.0100000, 0.0230506, 0.0531329, 0.1224745, 0.2823108, 0.5000000, 0.6507428, 1.5000000");
		index_2("0.0000000, 0.5000000, 1.0000000");
		values("0.0000000e+00, 8.3333333e-03, 1.6666667e-02", \
		  "0.0000000e+00, 1.9208818e-02, 3.8417635e-02", \
		  "0.0000000e+00, 4.4277440e-02, 8.8554881e-02", \
		  "0.0000000e+00, 1.0206207e-01, 2.0412415e-01", \
		  "0.0000000e+00, 2.3525901e-01, 4.7051801e-01", \
		  "0.0000000e+00, 4.1666667e-01, 8.3333333e-01", \
		  "0.0000000e+00, 5.4228568e-01, 1.0845714e+00", \
		  "0.0000000e+00, 1.2500000e+00, 2.5000000e+00");
	}
```

mithro, 2020-07-02:
@msaligane - In https://github.com/google/skywater-pdk/issues/37 you mentioned
> The variable comes before the index:

But the example copied from your comment at https://github.com/google/skywater-pdk/issues/37#issuecomment-653165896 above has exactly that ordering of the output?

msaligane, 2020-07-02:
That was the expected format that I pasted there (I know confusing..)

mithro, 2020-07-02:
@msaligane What is the *current* format? The index coming before variable?

msaligane, 2020-07-02:
@mithro Please look above:

What I pasted up here is the correct format:
```
lu_table_template ("del_1_7_7") {
		variable_1 : "input_net_transition";
		index_1("1, 2, 3, 4, 5, 6, 7");
		variable_2 : "total_output_net_capacitance";
		index_2("1, 2, 3, 4, 5, 6, 7");
```

Meaning that I am expecting variable before index. Maybe I can take a look at the code?

kkiningh, 2020-07-02:
This is the suggested format for the lu_table_template group.
```
library (name_string) {
 ...
 lu_table_template(name_string) {
  variable_1 : value_enum ;
  variable_2 : value_enum ;
  variable_3 : value_enum ;
  index_1 ("float, ..., float");
  index_2 ("float, ..., float");
  index_3 ("float, ..., float");
 }
}
```
In other words, all variables should come before all indices.

kkiningh, 2020-07-02:
Actually, looking closer it looks like the more general rule is: simple attributes first in alphabetical order (i.e. `key : value` pairs) and then complex attributes like `index` and `values`

msaligane, 2020-07-02:
@mithro Did you address the `normalized_driver_waveform` ? Primetime is flagging this:
Error message:
`Error: Line 24721, Cell 'sky130_fd_sc_hd__a311oi_2', The normalized_driver_waveform group with name 'ramp', has not been defined in the library. (LBDB-751)`

This is in the current .lib:

```
 normalized_driver_waveform ("driver_waveform_template") {
        index_1("0.0100000000, 0.0230506000, 0.0531329000, 0.1224745000, 0.2823108000, 0.5000000000, 0.6507428000, 1.5000000000");
        index_2("0.0000000000, 0.5000000000, 1.0000000000");
        driver_waveform_name : "ramp";
        values(            "0.0000000000, 0.0083333333, 0.0166666670", \
            "0.0000000000, 0.0192088180, 0.0384176350", \
            "0.0000000000, 0.0442774400, 0.0885548810", \
            "0.0000000000, 0.1020620700, 0.2041241500", \
            "0.0000000000, 0.2352590100, 0.4705180100", \
            "0.0000000000, 0.4166666700, 0.8333333300", \
            "0.0000000000, 0.5422856800, 1.0845714000", \
            "0.0000000000, 1.2500000000, 2.5000000000");
```

 And this is how the `normalized_driver_waveform` should be defined:

```
normalized_driver_waveform(waveform_template_name) {
 **driver_waveform_name** : string; /* Specifies the name of the driver
 waveform table */
 index_1 ("float…, float"); /* Specifies input net transition */
 index_2 ("float…, float"); /* Specifies normalized voltage */
 values ( "float…, float", \ /* Specifies the time in library units */
 …, \
 "float…, float");
 }
```

You have to reorder this to match the doc description.

mithro, 2020-07-02:
This sort order is currently defined at https://github.com/google/skywater-pdk/pull/34/files#diff-f9cb759bb20be5053e4a63e7e0b282d2R355-R372