google/skywater-pdk #45: Lint Python code to find syntax errors and undefined names

cclauss, 2020-07-02:
As discussed at https://github.com/google/skywater-pdk/pull/42#issuecomment-653190848, would it be useful to add a linting process to Travis CI or GitHub Actions to discover syntax errors and undefined names in Python code?

mithro, 2020-07-02:
I'm very supportive of adding flake8 and think it should be checked on probably Travis CI. At the moment Travis CI just checks you have updated `README.rst` from `README.src.rst` -- https://github.com/google/skywater-pdk/blob/master/.travis.yml#L27-L30

Generally we have used yapf rather than black for auto-formatting and would probably want to keep this consistent with the SymbiFlow projects - https://github.com/SymbiFlow

I'm open to exploring GitHub actions but have not used it in this capacity before.

cclauss, 2020-07-02:
I could add to either Travis CI or GitHub Actions.  Please let me know which way you want to go.

mithro, 2020-07-02:
@cclauss We should probably have a `make check` or similar type option in the Makefile first and then enable that checking on Travis CI / GitHub Actions?

Do any of the new GitHub actions do fancy things like publishing the output into the pull request as comments? That would be super cool to see.....

FYI -- For a consistent cross-platform Python 3 environment we are using Conda and will eventually use the EDDA packages (https://bit.ly/edda-conda-eda-spec).

PR #60 (2020-07-09): Makefile: Add a make lint-python command
Fixes #45

- [ ] Tests pass
- [ ] Appropriate changes to README are included in PR

If `make lint-python` is not run before `make env` then it fails to complete in 10 minutes which times out Travis CI.
commit e4f49b51 (2020-07-09): Makefile: Add a make lint-python command

Fixes #45
closed 2020-07-09 by mithro