google/skywater-pdk #47: Missing techfiles

kkiningh, 2020-07-02:
I'm trying to use the pdk with Synopsys tools (DC/ICC2), but I don't see a vendor or interconnect techfile (*.tf and *.itf respectively) which is needed for place and route and parasitics extraction. Will those be added later or is there another way I can obtain them?

mithro, 2020-07-02:
@kkiningh I don't believe anyone has used the PDK with th Synopsys tooling. 

Is a `.tf` / `.itf` file the same as a "tech lef" file (`.tlef`)? If so, I will be publishing them shortly too.

kkiningh, 2020-07-02:
I think the `.tf` can be generated from the `.tlef` but I'll have to look into how to do it.

mithro, 2020-07-06:
Initial `.tlef` files where added in https://github.com/google/skywater-pdk/commit/4cbdbb5a735a8acd53c6247eb2bc23b9245329f8

mithro, 2020-07-06:
@kkiningh - See also https://github.com/google/skywater-pdk/issues/62

20Mhz, 2020-07-07:
I believe by tf you mean Milkyway tech file, information may overlap with TECH tlef. For parasitics you will still be missing the ITF file, needed to generate TLUplus used as parasitic lookup tables. Maybe you can try Wire Load Model to start with.

kkiningh, 2020-07-07:
@20Mhz, yeah that's correct. I believe Milkyway supports a way to create a Milkyway library from a tlef and then exporting a techfile, although I've never tried it. You're right that the itf (from which the tluplus file can be generated) will still be missing.

@mithro I'm super busy for the next couple days, but I'll try to get around to trying to do this and report back soon.

20Mhz, 2020-07-07:
I believe LEF does not have enough info to automatically generate a mw tech. A fair work around could be to get an OpenAccess db and use Custom Compiler exporting capability to produce a mw tech as a starting point.

mithro, 2020-07-09:
FYI - Full technology LEF files for the standard cells where merged in https://github.com/google/skywater-pdk/pull/68

rovinski, 2020-07-09:
@mithro tech LEF is generally "good enough" for use in place-and-route tools for older nodes, e.g. 130nm. However, generally for timing signoff, designers would want to run a signoff-quality parasitic extraction which includes coupling capacitance. `.itf`, `.qrcTechfile`, etc. These checks are important for final power signoff, signal integrity signoff, and timing signoff.

20Mhz, 2020-07-10:
I'm now blocked by missing tech to GDS mapping.

mithro, 2020-07-10:
FYI - There is a spreadsheet at https://docs.google.com/spreadsheets/d/176OcP2h_TT_pDP2-zBLijlgFUP38LVgMQPlDtpLcrsk/edit which includes a lot of information about the layers.

@RTimothyEdwards has a bunch of useful resources at https://github.com/RTimothyEdwards/open_pdks (which will be eventually merged into this repository).

MohamedAliYounis, 2022-03-05:
@mithro could you please tell me how to get .lib files from .lib.json?