google/skywater-pdk #50: How is this supposed to work?

trcwm, 2020-07-03:
Cloned the repo, did 'make', got over 100MB of virtual environment installed.

typing 'conda activate skywater-pdk-scripts' says it can't find conda.

How do I get LEF/LIB/GDS2 files? 

```
Unpacking payload ...
Collecting package metadata (current_repodata.json): done                                         
Solving environment: done

## Package Plan ##

  environment location: /home/trc/programming/skywater-pdk/env/conda

  added / updated specs:
    - _libgcc_mutex==0.1=main
    - ca-certificates==2020.1.1=0
    - certifi==2020.4.5.1=py37_0
    - cffi==1.14.0=py37he30daa8_1
    - chardet==3.0.4=py37_1003
    - conda-package-handling==1.6.1=py37h7b6447c_0
    - conda==4.8.3=py37_0
    - cryptography==2.9.2=py37h1ba5d50_0
    - idna==2.9=py_1
    - ld_impl_linux-64==2.33.1=h53a641e_7
    - libedit==3.1.20181209=hc058e9b_0
    - libffi==3.3=he6710b0_1
    - libgcc-ng==9.1.0=hdf63c60_0
    - libstdcxx-ng==9.1.0=hdf63c60_0
    - ncurses==6.2=he6710b0_1
    - openssl==1.1.1g=h7b6447c_0
    - pip==20.0.2=py37_3
    - pycosat==0.6.3=py37h7b6447c_0
    - pycparser==2.20=py_0
    - pyopenssl==19.1.0=py37_0
    - pysocks==1.7.1=py37_0
    - python==3.7.7=hcff3b4d_5
    - readline==8.0=h7b6447c_0
    - requests==2.23.0=py37_0
    - ruamel_yaml==0.15.87=py37h7b6447c_0
    - setuptools==46.4.0=py37_0
    - six==1.14.0=py37_0
    - sqlite==3.31.1=h62c20be_1
    - tk==8.6.8=hbc83047_0
    - tqdm==4.46.0=py_0
    - urllib3==1.25.8=py37_0
    - wheel==0.34.2=py37_0
    - xz==5.2.5=h7b6447c_0
    - yaml==0.1.7=had09818_2
    - zlib==1.2.11=h7b6447c_3

The following NEW packages will be INSTALLED:

  _libgcc_mutex      pkgs/main/linux-64::_libgcc_mutex-0.1-main
  ca-certificates    pkgs/main/linux-64::ca-certificates-2020.1.1-0
  certifi            pkgs/main/linux-64::certifi-2020.4.5.1-py37_0
  cffi               pkgs/main/linux-64::cffi-1.14.0-py37he30daa8_1
  chardet            pkgs/main/linux-64::chardet-3.0.4-py37_1003
  conda              pkgs/main/linux-64::conda-4.8.3-py37_0
  conda-package-han~ pkgs/main/linux-64::conda-package-handling-1.6.1-py37h7b6447c_0
  cryptography       pkgs/main/linux-64::cryptography-2.9.2-py37h1ba5d50_0
  idna               pkgs/main/noarch::idna-2.9-py_1
  ld_impl_linux-64   pkgs/main/linux-64::ld_impl_linux-64-2.33.1-h53a641e_7
  libedit            pkgs/main/linux-64::libedit-3.1.20181209-hc058e9b_0
  libffi             pkgs/main/linux-64::libffi-3.3-he6710b0_1
  libgcc-ng          pkgs/main/linux-64::libgcc-ng-9.1.0-hdf63c60_0
  libstdcxx-ng       pkgs/main/linux-64::libstdcxx-ng-9.1.0-hdf63c60_0
  ncurses            pkgs/main/linux-64::ncurses-6.2-he6710b0_1
  openssl            pkgs/main/linux-64::openssl-1.1.1g-h7b6447c_0
[... 131 lines cut ...]
Building wheels for collected packages: sphinx-symbiflow-theme, sphinxcontrib-verilog-diagrams
  Building wheel for sphinx-symbiflow-theme (setup.py): started
  Building wheel for sphinx-symbiflow-theme (setup.py): finished with status 'done'
  Created wheel for sphinx-symbiflow-theme: filename=sphinx_symbiflow_theme-0.1.11-py3-none-any.whl size=18259084 sha256=e231179dd34d4172f2bbee0c01b0bf9d78105c67241fb0242bf3e41c8b7adc6e
  Stored in directory: /tmp/pip-ephem-wheel-cache-rxdt1o5p/wheels/b3/01/48/cc3bca817efc8c3dcca37593870c5a6508ec9d22294300a880
  Building wheel for sphinxcontrib-verilog-diagrams (setup.py): started
  Building wheel for sphinxcontrib-verilog-diagrams (setup.py): finished with status 'done'
  Created wheel for sphinxcontrib-verilog-diagrams: filename=sphinxcontrib_verilog_diagrams-0.0.dev0-py3-none-any.whl size=12122 sha256=8364a9e9fbd5a934af3306d00aa43bde17a3013cb0434f19ed8cd71487cd99d2
  Stored in directory: /tmp/pip-ephem-wheel-cache-rxdt1o5p/wheels/54/62/3b/9549f1fdbdc9a36463b47d2c75617b4cc43f2c626a6e2ca5bb
Successfully built sphinx-symbiflow-theme sphinxcontrib-verilog-diagrams
Installing collected packages: typing, attrs, wcwidth, more-itertools, pluggy, six, pyparsing, packaging, py, pytest, rst-include, sphinx-symbiflow-theme, docutils, idna, chardet, urllib3, requests, sphinxcontrib-serializinghtml, sphinxcontrib-htmlhelp, snowballstemmer, pytz, babel, imagesize, Pygments, alabaster, sphinxcontrib-jsmath, sphinxcontrib-devhelp, sphinxcontrib-applehelp, sphinxcontrib-qthelp, MarkupSafe, Jinja2, sphinx, argh, pathtools, tornado, watchdog, livereload, PyYAML, port-for, sphinx-autobuild, sphinxcontrib-verilog-diagrams, hdlparse, symbolator, restructuredtext-lint, tabulate, stringcase, marshmallow, marshmallow-enum, typing-extensions, mypy-extensions, typing-inspect, dataclasses-json, skywater-pdk
  Running setup.py develop for skywater-pdk
Successfully installed Jinja2-2.11.2 MarkupSafe-1.1.1 PyYAML-5.3.1 Pygments-2.6.1 alabaster-0.7.12 argh-0.26.2 attrs-19.3.0 babel-2.8.0 chardet-3.0.4 dataclasses-json-0.5.1 docutils-0.16 hdlparse-1.0.4 idna-2.10 imagesize-1.2.0 livereload-2.6.2 marshmallow-3.6.1 marshmallow-enum-1.5.1 more-itertools-8.4.0 mypy-extensions-0.4.3 packaging-20.4 pathtools-0.1.2 pluggy-0.13.1 port-for-0.3.1 py-1.9.0 pyparsing-2.4.7 pytest-5.4.3 pytz-2020.1 requests-2.24.0 restructuredtext-lint-1.3.1 rst-include-1.0.8 six-1.15.0 skywater-pdk snowballstemmer-2.0.0 sphinx-3.1.1 sphinx-autobuild-0.7.1 sphinx-symbiflow-theme-0.1.11 sphinxcontrib-applehelp-1.0.2 sphinxcontrib-devhelp-1.0.2 sphinxcontrib-htmlhelp-1.0.3 sphinxcontrib-jsmath-1.0.1 sphinxcontrib-qthelp-1.0.3 sphinxcontrib-serializinghtml-1.1.4 sphinxcontrib-verilog-diagrams-0.0.dev0 stringcase-1.2.0 symbolator-1.0.2 tabulate-0.8.7 tornado-6.0.4 typing-3.7.4.1 typing-extensions-3.7.4.2 typing-inspect-0.6.0 urllib3-1.25.9 watchdog-0.10.3 wcwidth-0.2.5

#
# To activate this environment, use
#
#     $ conda activate skywater-pdk-scripts
#
# To deactivate an active environment, use
#
#     $ conda deactivate

touch /home/trc/programming/skywater-pdk/env/conda/envs/skywater-pdk-scripts/bin/python
```

kkiningh, 2020-07-03:
The actual GDS and LEF files are being added in submodules under the library folder in pull request #33 . In the mean time, you can get them from [here](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/).

mithro, 2020-07-03:
I started the pull request and then got distracted responding to all the emails that the talk generated. That should be merged in the next day or two but you can look at the WIP stuff as pointed out by @kkiningh.

mithro, 2020-07-09:
The pull request has been merged and you will now get the files if you clone the master branch.

closed 2020-07-09 by mithro

Amjad2004, 2020-08-26:
The actual GDS and LEF files are being added in submodules under the libraries folder in pull request #35 In the mean time, you can get them from here