google/skywater-pdk #51: Publish script which combines the LEF files from the cells into one big file

mithro, 2020-07-03:
Similar to how there is a script which combines the timing data into a single large JSON file. We need a script which does the same for the LEF files.

msaligane, 2020-07-03:
@mithro I made one but it is pretty rudimentary. Do you want me to push it?

mithro, 2020-07-03:
@msaligane Sure! It should go into the Python API library at `skywater-pdk/scripts/python-skywater-pdk`

RTimothyEdwards, 2020-07-04:
@mithro : I have this capability as part of the open_pdks "foundry install" script.  I would also refer to it as "rudimentary".  But, honestly, a library of LEF macros is little more than writing one single header and then dumping all the macros.  About the only complexity is to make sure it isn't dumping the same macro more than once.

mithro, 2020-07-05:
@msaligane As published a LEF combinding script at https://github.com/mithro/skywater-pdk/pull/1/files

mithro, 2020-07-27:
FYI - [The open_pdk repo has a `create_lef_library.py`](https://github.com/RTimothyEdwards/open_pdks/blob/master/common/create_lef_library.py) which should be imported into the skywater-pdk Python API library and used here.