google/skywater-pdk #52: Publish script which combines the GDS files from the cells into one big file

mithro, 2020-07-03:
Similar to how there is a script which combines the timing data from the per cell timing JSON data into a single large liberty file, we need a script which does the same for the GDS files.

As GDS files are a binary format, this script should use Magic. (Magic should be installed into the environment from conda from https://github.com/litex-hub/litex-conda-eda/tree/master/misc/magic). This can be done by generating a TCL script to give to Magic.

mithro, 2020-07-03:
From @RTimothyEdwards;
> The steps needed are;
> 1. Read all the individual GDS files
> 2. Create a new cell with the name of the library
> 3. Instantiate one of every individual GDS cell in the new parent cell
> 4. Write GDS of the parent cell

RTimothyEdwards, 2020-07-04:
@mithro :  I have this capability as part of the open_pdks "foundry install" script.  It follows the steps you (or rather I) outlined above, although with additional technical details.  I think it would be helpful for me to add an option to the "gds write" command in magic which would prevent it from writing the top-level cell, so that the result would be just a GDS library of components.  The additional top-level cell is unnecessary, and while it should not cause problems, it could potentially confuse applications that read the GDS library.

mithro, 2020-07-05:
https://github.com/RTimothyEdwards/open_pdks/blob/master/common/foundry_install.py#L2342-L2343

msaligane, 2020-07-22:
@wenbodd Can you try this and merge the gds files for openroad?

wenbodd, 2020-07-22:
The merged gds is pushed to the sky130 branch in openroad:
[https://github.com/The-OpenROAD-Project/OpenROAD-flow/tree/sky130/flow/platforms/sky130/gds](url)

mithro, 2020-07-27:
FYI - [The open_pdk repo has a `create_gds_library.py`](https://github.com/RTimothyEdwards/open_pdks/blob/master/common/create_gds_library.py) which should be imported into the skywater-pdk Python API library and used here.