google/skywater-pdk #53: Add documentation on how to use the SkyWater PDK on the efabless platform

mithro, 2020-07-04:
It would be good to have a how to in the documentation around signing up to the efabless platform and getting started with the open source tools on the SkyWater PDK. 

efabless provides machines which are pre-setup including all the open source tooling and this PDK. They also have examples, so it is a very easy way to get started.

RTimothyEdwards, 2020-07-04:
@mithro :  We do not have the PDK on the efabless platform.  We have a preliminary PDK that I built around the original SkyWater files that is on the development server (staff access only).  It will need to be redone from the Google/SkyWater repository.  I have been waiting for the primitive devices and device models, without which the PDK is useless for anything other than digital synthesis.  I can go ahead with that part of the PDK, but I am still waiting for something to show up in the repo regarding the MOST BASIC INSTRUCTIONS for installing the PDK.  As it stands, after cloning the skywater-pdk repo, I see a lot of stuff about what the PDK is, but nothing about how to get it onto a local computer and do something useful with it.

RTimothyEdwards, 2020-07-04:
@mithro :  In short, I'm waiting for standard cells to appear in the repository along with some instructions on how to get all those files to show up on my computer.  I have not seen either yet.

mithro, 2020-07-04:
As mentioned on IRC, at the moment you need to add use my fork to see work in progress changes.

The process for getting changes into the repository is;
 1) Create your changes
 2) Send pull request and have people review the pull request
 3) Have the pull request approved and merged

To review a pull request you need to be able to view the work in progress changes.

Here are some helpful examples;

#### Clone the upstream repository

```
git clone git+ssh://github.com/google/skywater-pdk.git
```

#### Add my fork as a remote

```
cd skywater-pdk
git remote add mithro git+ssh://github.com/mithro/skywater-pdk.git
git fetch mithro
```

#### Checkout a branch and review

```
git checkout mithro/liberty -b liberty
git submodule update --init
make timing
```

#### Useful links explaining this common workflow

 * https://gist.github.com/Chaser324/ce0505fbed06b947d962
 * https://guides.github.com/activities/forking/
 * https://reflectoring.io/github-fork-and-pull/
 * https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/about-collaborative-development-models

RTimothyEdwards, 2020-07-04:
@mithro :  Let's just assume for the moment that I, like most people in the world except for a handful of software developers, don't understand all this freaking git pull/fork/clone/remote/fetch/head/origin crap, and that I am barely able to manage to work with git most of the time by having a text file with a few sets of instructions on what to do in some situations, and that when everything gets into a hopeless state, I just delete everything and re-clone.  Also, let's assume that I am not going to make any progress on my part of the project until the repository is in a state such that I can make a simple set of instructions to follow that will get me files on my computer that I can work with to build out an organized PDK for use with magic/qflow/netgen/etc.

RTimothyEdwards, 2020-07-04:
https://xkcd.com/1597/