google/skywater-pdk #54: Publish script which combines the CDL files from the cells (and models) into one big file

mithro, 2020-07-04:
Similar to how there is a script which combines the timing data from the per cell timing JSON data into a single large liberty file, we need a script which does the same for the CDL files. This also needs to make sure to include the models.

The script should be added to the [`skywater_pdk` Python module](https://github.com/google/skywater-pdk/tree/master/scripts/python-skywater-pdk).

RTimothyEdwards, 2020-07-04:
@mithro :  I have this capability as part of the open_pdks "foundry install" script.

mithro, 2020-07-27:
FYI - [The open_pdk repo has a `create_spice_library.py`](https://github.com/RTimothyEdwards/open_pdks/blob/master/common/create_spice_library.py) which should be imported into the skywater-pdk Python API library and used here.