google/skywater-pdk #59: SPICE model

omranh, 2020-07-05:
If I just want to download the MOSFET SPICE model, how can I get it?

lhrodovalho, 2020-07-05:
I'm also looking for the spice models. They would be an initial step to design any analog circuits before we can actually use the PDK to make the layouts.

mithro, 2020-07-05:
@omranh / @lhrodovalho - The data required to do analog circuit design has not yet been released, but will hopefully be released before the end of the month.

agustincampeny, 2020-08-09:
Is there any news on when will the models be uploaded. We are looking on using this process for an analog circuit design course, since ON Semi 0.5um is rather old.

omranh, 2020-08-30:
Any update on this?

mithro, 2020-09-25:
Spice models are currently available at https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_pr/ and will be merged into this repo in the near future.

There is some early documentation around the primitives in the pull request at https://github.com/google/skywater-pdk/pull/136 which is currently visible at https://skywater-pdk--136.org.readthedocs.build/en/136/rules/device-details.html -- when merged it will appear at https://skywater-pdk.rtfd.io with all the other current documentation.

omranh, 2020-10-03:
Many thanks for the great effort and the very nice documentation. Do you have Monte Carlo process/mismatch models as well?

mithro, 2020-10-05:
@omranh Yes, but it is unclear how to use the data with ngspice. @RTimothyEdwards is looking into it further.

mithro, 2020-10-05:
BTW The pull request #136 has been merged into the repository and the docs now appear at https://skywater-pdk.rtfd.io

omranh, 2020-10-05:
What about MC models for Cadence Spectre?

closed 2020-10-05 by mithro