google/skywater-pdk #66: Validating GDS files

cwgreene, 2020-07-08:
Hi! So there is documentation that says what the process design rules are

https://skywater-pdk.readthedocs.io/en/latest/rules.html

Are consumers of the PDK expected to manually convert these rules into design rules that let a GDS file be checked using a program like Magic? Are these rules somewhere in the checkout directories that are already in a format that can be automatically processed? Or is there a tool in the checkout that can be used to validate a GDS file against the design rules?

mithro, 2020-07-08:
@cwgreene -- The design rules will be published in computer parsable documentation form in the (hopefully near) future. 

The PDK will also include in the future;
 * a Magic configuration which checks all the design rules -- Work in progress versions of these files can be downloaded from @RTimothyEdwards 's open_pdks repo at https://github.com/RTimothyEdwards/open_pdks
 * a Python script which generates a configuration suitable to be loaded into Mentor Calibre

We are also looking at setting up a service with efabless which lets you submit GDS to be verified to be DRC clean. This is needed for the open source MPW shuttle service, so should also be available soon.

mithro, 2020-07-08:
FYI - A lot of the source of the design rule documentation comes from CSV files which can be found at https://github.com/google/skywater-pdk/tree/master/docs/rules/periphery and https://github.com/google/skywater-pdk/tree/master/docs/rules/assumptions

cwgreene, 2020-07-08:
Ah thanks! That explains a lot! Question; is the information in the csv files sufficient to derive the full rules than that would be fed to something like magic? Are `.tlef` files a standard way of representing these rules?

mithro, 2020-07-08:
@cwgreene -- Not currently but hopefully will in the future.