google/skywater-pdk #69: sky130_fd_sc_hvl : The 'clk_width' attribute, which expects values of string type

msaligane, 2020-07-09:
## Actual Behavior
The following error happens when reading the libs in primetime:
```
Error: Line 36, The 'clk_width' attribute, which expects values of string type,
    is being supplied a value of floating-point type. (LBDB-53)
clk_width : 2.500000e-08;
```

## Expected Behavior
To fix this issue:
clk_width should be defined this way:
` clk_width : "2.500000e-08";
`

mithro, 2020-07-09:
@msaligane -- Wow, that is kind of stupid....

mithro, 2020-07-09:
@msaligane - This affects the liberty files for all the digital cells, right?

msaligane, 2020-07-09:
@mithro I only see this in the hvl library (maybe because the corners are different) but yes for the digital cells.

commit b70d1a51 (2020-07-15): scripts/liberty: Make the `clk_width` attribute a string.

Fixes #69.
closed 2020-07-15 by mithro