google/skywater-pdk #72: sky130_fd_sc_hvl: The 'tran_timestep' attribute, which expects values of string type

msaligane, 2020-07-09:
## Actual Behavior

The following error message is reported when reading this lib in primetime:
```
Error: Line 268, The 'tran_timestep' attribute, which expects values of string type,
    is being supplied a value of floating-point type. (LBDB-53)
tran_timestep : 1.000000e-12; 

```

## Expected Behavior
The correct syntax is:
`tran_timestep : "1.000000e-12";
`

mithro, 2020-07-10:
The issue seems to be that the `define ("attribute_name", "group_name", "attribute_type") ;` at the top of the file is using `attribute_type: string` rather than `attribute_type: float`.

```json
  "define": [
    {
      "attribute_name": "switching_power_split_model",
      "attribute_type": "string",
      "group_name": "library"
    },
    {
      "attribute_name": "driver_model",
      "attribute_type": "string",
      "group_name": "library"
    },
    {
      "attribute_name": "tran_timestep",
      "attribute_type": "string",
      "group_name": "library"
    },
    {
      "attribute_name": "leakage_sim_opt",
      "attribute_type": "string",
      "group_name": "library"
    },
    {
      "attribute_name": "default_arc_mode",
      "attribute_type": "string",
      "group_name": "library"
    },
    {
      "attribute_name": "def_sim_opt",
      "attribute_type": "string",
      "group_name": "library"
    },
    {
      "attribute_name": "simulator",
      "attribute_type": "string",
      "group_name": "library"
    },
    {
      "attribute_name": "violation_delay_degrade_pct",
      "attribute_type": "string",
      "group_name": "timing"
    },
    {
      "attribute_name": "sim_opt",
      "attribute_type": "string",
      "group_name": "timing"
    }
  ],
```

msaligane, 2020-07-12:
This is fixed I think.

closed 2020-07-12 by msaligane