google/skywater-pdk #74: Memory Generator and IO Cells

leoheck, 2020-07-09:
Hi, @mithro I am following your hard work here, and I don't want to bother you much but I was reading the README, and I couldn't find any mention of a memory generator and also of IO Cells. Do you know if this project will have such features available too?

mithro, 2020-07-09:
There will be both IO cells (https://github.com/google/skywater-pdk/issues/5) and a both an SRAM (https://github.com/google/skywater-pdk/issues/3) and flash (https://github.com/google/skywater-pdk/issues/4) build spaces released in the future. The build spaces need to be used with a memory generator to actually create usable blocks however.

The SRAM build space will be published with support for the [OpenRAM memory generator](https://openram.soe.ucsc.edu/) and eventually contain a bunch of precompiled macros for common configurations. @mguthaus will be giving a talk about OpenRAM as part of the [FOSSi Foundation's Dial Up talk series](https://fossi-foundation.org/dial-up/).

This is all documented in the GitHub issue tracker and the [Known Issues page](https://skywater-pdk.readthedocs.io/en/latest/known_issues.html).

Would love to have a pull request to add stuff to the README if you think that makes sense.

closed 2020-07-09 by mithro

leoheck, 2020-07-09:
> Would love to have a pull request to add stuff to the README if you think that makes sense.

Oh, that is interesting! I am not totally into this project to be able to add something useful, but since you gave me a good and informative reply I will try that.