google/skywater-pdk #75: Periphery rules diagrams are referencing rules not in text

riking, 2020-07-09:
## Expected Behavior
All rule numbers referenced in diagrams are present in the document.

## Actual Behavior
Several rule sections have the final rule cut off.

E.g. the photodiode section only goes up to `photo.10` but the image references `photo.11`

https://skywater-pdk.readthedocs.io/en/latest/rules/periphery.html#photo

https://skywater-pdk.readthedocs.io/en/latest/_images/p031-photo_dotdash.svg

mithro, 2020-07-09:
Thanks for the report! I will investigate. If you have time, it would be super useful if you could provide a list of these missing rule numbers that you have found.

RTimothyEdwards, 2020-07-09:
@mithro :  I can confirm that this is nearly ubiquitous.  Just checking the first ten or so rules, I think only one of them was not missing the last rule.  For example, these rules that are in the original SkyWater DRC PDF document are missing:  dnwell:7, pwbm:5, pwdem:6, hvtp:6, hvtr:3, lvtn:14, ncm:8.  So it is clearly a bug in the script that is losing the last line of nearly every rule set.

mithro, 2020-07-09:
Probably cause by the https://github.com/google/skywater-pdk/blob/master/docs/rules/periphery/periphery-split-csv.py script then?

mithro, 2020-07-09:
Looks like the csv file is also missing the last line? https://github.com/google/skywater-pdk/blob/master/docs/rules/periphery/p031-photo_dotdash.csv

mithro, 2020-07-09:
Does look like that rule is in the base csv file @ https://github.com/google/skywater-pdk/blob/master/docs/rules/periphery/periphery.csv#L541 -- so the `periphery-split-csv.py` script is most definately the issue.

PR #147 (2020-10-16): docs: Fix final rule being incorrectly excluded
Fixes #75

Prior behavior was that the final element of the `rows` list would be removed for _every_ Rule in the CSV file when the rule did not contain a "Use"+"Explanation" section.  This PR puts a sanity check on that logic so that we only do it we actually need to (i.e. when a "Use"+"Explanation" section is present in the rule).
commit ec448b28 (2020-10-16): Fix #75 - final rule was excluded incorrectly
commit fb266d10 (2020-10-16): Merge pull request #147 from wanderrful/fix-periphery-rules

docs: Fix final rule being incorrectly excluded

Fixes #75

Prior behavior was that the final element of the `rows` list would be removed for _every_ Rule in the CSV file when the rule did not contain a "Use"+"Explanation" section.  This PR puts a sanity check on that logic so that we only do it we actually need to (i.e. when a "Use"+"Explanation" section is present in the rule).
closed 2020-10-16 by mithro