google/skywater-pdk #77: none _ccsnoise liberty files containing _ccsnoise info for hs and ms libraries

agorararmard, 2020-07-09:
## Expected Behavior
sky130_fd_sc_hs__tt_025C_1v80.lib shouldn't contain ccsnoise related info. this is also the case for sky130_fd_sc_ms__tt_025C_1v80.lib

## Actual Behavior
sky130_fd_sc_hs__tt_025C_1v80.lib contains ccsnoise related info: i.e.
for inv_2:
`             ccsn_first_stage () {
                    dc_current ("ccsn_dc") {
                        index_1("-1.800000000, -0.900000000, -0.360000000, -0.180000000, 0.0000000000, 0.0900000000, 0.1800000000, 0.2700000000, 0.3600000000, 0.4500000000, 0.5400000000, 0.6300000000, 0.7200000000, 0.8100000000, 0.9000000000, 0.9900000000, 1.0800000000, 1.1700000000, 1.2600000000, 1.3500000000, 1.4400000000, 1.5300000000, 1.6200000000, 1.7100000000, 1.8000000000, 1.9800000000, 2.1600000000, 2.7000000000, 3.6000000000");
                        index_2("-1.800000000, -0.900000000, -0.360000000, -0.180000000, 0.0000000000, 0.0900000000, 0.1800000000, 0.2700000000, 0.3600000000, 0.4500000000, 0.5400000000, 0.6300000000, 0.7200000000, 0.8100000000, 0.9000000000, 0.9900000000, 1.0800000000, 1.1700000000, 1.2600000000, 1.3500000000, 1.4400000000, 1.5300000000, 1.6200000000, 1.7100000000, 1.8000000000, 1.9800000000, 2.1600000000, 2.7000000000, 3.6000000000");
                        values(                            "8.8354000000, 1.3865400000, 1.0756500000, 1.0189900000, 0.9559030000, 0.9219560000, 0.8864230000, 0.8493210000, 0.8106690000, 0.7704870000, 0.7287960000, 0.6856180000, 0.6409750000, 0.5948900000, 0.5473880000, 0.4984930000, 0.4482300000, 0.3966250000, 0.3437030000, 0.2894910000, 0.2340150000, 0.1773010000, 0.1193760000, 0.0602669000, -5.29305e-09, -0.119204000, -0.235303000, -0.646946000, -3.527050000", \
                            "8.5783400000, 1.1798400000, 0.9202040000, 0.8884130000, 0.8534060000, 0.8341080000, 0.8131440000, 0.7901050000, 0.7646560000, 0.7366030000, 0.7058640000, 0.6724120000, 0.6362450000, 0.5973740000, 0.5558190000, 0.5116040000, 0.4647600000, 0.4153180000, 0.3633170000, 0.3087960000, 0.2517990000, 0.1923720000, 0.1305620000, 0.0664210000, -2.20984e-08, -0.133245000, -0.264334000, -0.719539000, -3.547680000", \
                            "8.2941900000, 0.9276520000, 0.6915660000, 0.6703380000, 0.6480140000, 0.6362970000, 0.6240940000, 0.6112750000, 0.5976580000, 0.5829720000, 0.5668130000, 0.5486070000, 0.5276900000, 0.5035440000, 0.4759020000, 0.4446550000, 0.4097570000, 0.3711950000, 0.3289700000, 0.2830930000, 0.2335880000, 0.1804840000, 0.1238200000, 0.0636412000, -3.42017e-09, -0.131318000, -0.264331000, -0.731207000, -3.556270000", \
                            "8.1947800000, 0.8343000000, 0.5962890000, 0.5780840000, 0.5591240000, 0.5492730000, 0.5391060000, 0.5285510000, 0.5175070000, 0.5058300000, 0.4933050000, 0.4796040000, 0.4642060000, 0.4463770000, 0.4253640000, 0.4006920000, 0.3721480000, 0.3396440000, 0.3031380000, 0.2626130000, 0.2180670000, 0.1695070000, 0.1169550000, 0.0604394000, -3.37961e-09, -0.126859000, -0.257757000, -0.723358000, -3.555580000", \
                            "8.1048900000, 0.7688150000, 0.4955330000, 0.4800920000, 0.4641770000, 0.4559790000, 0.4475810000, 0.4389420000, 0.4300100000, 0.4207070000, 0.4109300000, 0.4005180000, 0.3892240000, 0.3766340000, 0.3620540000, 0.3445520000, 0.3233590000, 0.2980920000, 0.2685880000, 0.2347690000, 0.1965910000, 0.1540290000, 0.1070730000, 0.0557248000, -7.79901e-09, -0.119766000, -0.246572000, -0.707046000, -3.553310000", \
                            "8.0619800000, 0.7571000000, 0.4444100000, 0.4301240000, 0.4156160000, 0.4081730000, 0.4005740000, 0.3927890000, 0.3847790000, 0.3764910000, 0.3678500000, 0.3587490000, 0.3490230000, 0.3384040000, 0.3264190000, 0.3122530000, 0.2948810000, 0.2735710000, 0.2479910000, 0.2180000000, 0.1835210000, 0.1445040000, 0.1009200000, 0.0527529000, -4.02374e-08, -0.115120000, -0.239029000, -0.695213000, -3.551480000", \
                            "8.0195300000, 0.7592220000, 0.3941390000, 0.3802570000, 0.3670770000, 0.3603450000, 0.3534930000, 0.3465010000, 0.3393400000, 0.3319720000, 0.3243460000, 0.3163900000, 0.3079960000, 0.2989940000, 0.2890940000, 0.2777440000, 0.2639810000, 0.2467120000, 0.2252730000, 0.1993840000, 0.1689150000, 0.1337880000, 0.0939494000, 0.0493608000, -3.19743e-07, -0.109702000, -0.230097000, -0.680680000, -3.549140000", \
                            "7.9775400000, 0.7709770000, 0.3480860000, 0.3311490000, 0.3191790000, 0.3131150000, 0.3069640000, 0.3007080000, 0.2943290000, 0.2878010000, 0.2810890000, 0.2741430000, 0.2668960000, 0.2592410000, 0.2510030000, 0.2418650000, 0.2311710000, 0.2177860000, 0.2005970000, 0.1790280000, 0.1528440000, 0.1219210000, 0.0861783000, 0.0455523000, -2.68143e-06, -0.103504000, -0.219740000, -0.663290000, -3.546270000", \
                            "7.9364400000, 0.7878830000, 0.3147650000, 0.2836950000, 0.2725720000, 0.2671360000, 0.2616390000, 0.2560690000, 0.2504130000, 0.2446540000, 0.2387670000, 0.2327220000, 0.2264730000, 0.2199570000, 0.2130710000, 0.2056380000, 0.1973020000, 0.1872690000, 0.1742270000, 0.1571040000, 0.1354200000, 0.1089710000, 0.0776385000, 0.0413313000, -2.21030e-05, -0.096550400, -0.207969000, -0.642949000, -3.542930000", \
                            "7.8968900000, 0.8069130000, 0.3005910000, 0.2399950000, 0.2279210000, 0.2230570000, 0.2181710000, 0.2132380000, 0.2082500000, 0.2031930000, 0.1980540000, 0.1928130000, 0.1874410000, 0.1819000000, 0.1761310000, 0.1700370000, 0.1634410000, 0.1559350000, 0.1465430000, 0.1338040000, 0.1167430000, 0.0949711000, 0.0683061000, 0.0366253000, -0.000170862, -0.089002000, -0.194961000, -0.619808000, -3.539820000", \
                            "7.8596000000, 0.8267920000, 0.3002830000, 0.2062960000, 0.1858820000, 0.1814240000, 0.1770850000, 0.1727250000, 0.1683340000, 0.1639030000, 0.1594220000, 0.1548800000, 0.1502600000, 0.1455390000, 0.1406840000, 0.1356440000, 0.1303310000, 0.1245640000, 0.1178680000, 0.1090410000, 0.0965366000, 0.0795181000, 0.0576697000, 0.0308217000, -0.001155150, -0.081721500, -0.181707000, -0.595238000, -3.540550000", \
                            "7.8252600000, 0.8474990000, 0.3066080000, 0.1889820000, 0.1470810000, 0.1421300000, 0.1381130000, 0.1341470000, 0.1301780000, 0.1261920000, 0.1221780000, 0.1181300000, 0.1140360000, 0.1098830000, 0.1056540000, 0.1013220000, 0.0968430000, 0.0921410000, 0.0870339000, 0.0809529000, 0.0724460000, 0.0599642000, 0.0428284000, 0.0207766000, -0.006356470, -0.078536300, -0.172456000, -0.574884000, -3.556480000", \
                            "7.7944700000, 0.8696340000, 0.3166740000, 0.1825880000, 0.1120880000, 0.1024810000, 0.0974312000, 0.0931689000, 0.0890564000, 0.0849910000, 0.0809377000, 0.0768789000, 0.0728029000, 0.0686991000, 0.0645563000, 0.0603603000, 0.0560910000, 0.0517154000, 0.0471685000, 0.0422686000, 0.0363280000, 0.0276462000, 0.0147674000, -0.002835470, -0.025386400, -0.089542300, -0.177694000, -0.570116000, -3.601660000", \
                            "7.7677000000, 0.8939660000, 0.3299980000, 0.1803370000, 0.0813966000, 0.0596518000, 0.0480692000, 0.0413877000, 0.0360008000, 0.0310545000, 0.0263157000, 0.0216926000, 0.0171397000, 0.0126308000, 0.0081479200, 0.0036770500, -0.000795384, -0.005285580, -0.009818280, -0.014446000, -0.019348700, -0.025357000, -0.034384700, -0.047704500, -0.065708100, -0.121594000, -0.203525000, -0.584131000, -3.671170000", \
                            "7.7453300000, 0.9211730000, 0.3465950000, 0.1807340000, 0.0554120000, 0.0180132000, -0.005982570, -0.019189000, -0.027409200, -0.033929700, -0.039707000, -0.045077100, -0.050190400, -0.055125900, -0.059930000, -0.064632700, -0.069255100, -0.073813800, -0.078324500, -0.082806800, -0.087301800, -0.091985400, -0.097856000, -0.106959000, -0.120291000, -0.167160000, -0.241873000, -0.607079000, -3.749370000", \
                            "7.7275800000, 0.9517520000, 0.3663800000, 0.1841040000, 0.0344464000, -0.018160800, -0.055779400, -0.078961300, -0.091878200, -0.100657000, -0.107737000, -0.113941000, -0.119615000, -0.124930000, -0.129982000, -0.134831000, -0.139517000, -0.144067000, -0.148500000, -0.152834000, -0.157085000, -0.161286000, -0.165613000, -0.171289000, -0.180286000, -0.217784000, -0.284568000, -0.631581000, -3.828450000", \
                            "7.7145800000, 0.9859950000, 0.3891300000, 0.1908530000, 0.0187246000, -0.047718600, -0.098327500, -0.132803000, -0.152242000, -0.164038000, -0.172733000, -0.179913000, -0.186213000, -0.191936000, -0.197250000, - [... cut ...]
                            values("0.1966904000, 0.3147046400, 0.3933808000, 0.3147046400, 0.1966904000");
                        }
                    }
                    stage_type : "both";
                }`

## Steps to Reproduce the Problem

1. run make sky130_fd_sc_hs
2. check the timing folder under v0.0.2
3. You'll find that both sky130_fd_sc_hs__tt_025C_1v80.lib and sky130_fd_sc_hs__tt_025C_1v80_ccsnoise.lib contain the ccsnoise information.

This is also the case for the ms library

## Specifications

- Version: v0.0.2

mithro, 2020-07-13:
@agorararmard This should be fixed now.

agorararmard, 2020-07-13:
@mithro  I just pulled and did git submodule update --init and ran make sky130_fd_sc_hs. I found that the issue still exists in all three versions (latest, v0.0.1, v0.0.2). so to reiterate on what's the problem:

1- we have one json description for the typical hs library with 25C and 1.80V: sky130_fd_sc_hs__tt_025C_1v80_ccsnoise.lib.json
2- However, two liberties are generated: sky130_fd_sc_hs__tt_025C_1v80_ccsnoise.lib and sky130_fd_sc_hs__tt_025C_1v80.lib.
3- We at efabless, openLane team, are only interested in this one sky130_fd_sc_hs__tt_025C_1v80.lib
4- Because yosys+ABC doesn't know how to parse ccsnoise information from liberty, and that's why we expect the file that doesn't contain the _ccsnoise postfix not to contain any ccsnoice information. Thus, the flow fails at synthesis.

I can share with you a small design so you can run against yosys and validate it from your side.

mithro, 2020-07-13:
A number of other people have successfully use these `.lib` files with Yosys, so it is unclear to me what is going on.

Can you please explain both;
 * How the `sky130_fd_sc_hs__tt_025C_1v80.lib` files has ccsnoise in it? 
 * The error you are getting from Yosys?

agorararmard, 2020-07-13:
For any given cell in sky130_fd_sc_hs__tt_025C_1v80.lib there exists a timing section under which there exists: `ccsn_first_stage () ` as a sub section. this subsection shouldn't exist since sky130_fd_sc_hs__tt_025C_1v80 doesn't have the _ccsnoise postfix. The whole subsection is listed above. Also any other ccsn related subsection. 

As for the yosys crash, we're still investigating the root cause of this. However, it worked fine with the hd library and it doesn't have the same issue (the cssn subsections).

mithro, 2020-07-13:
It sounds like you are using an old version of the PDK? I just checked `sky130_fd_sc_hs__tt_025C_1v80.lib` and could find no `ccsn_first_stage ()` in the file?

mithro, 2020-07-13:
Please make sure you have https://github.com/google/skywater-pdk/pull/92 in your repo.

agorararmard, 2020-07-14:
@mithro 
this is my git log from a fresh clone
```
commit ad3b507e384ac364472571bcb1bc4e02fd649ad6 (HEAD -> master, origin/master, origin/HEAD)
Merge: f2838cf b626ca0
Author: Tim Ansell <me@mith.ro>
Date:   Sun Jul 12 22:59:42 2020 -0700

    Merge pull request #92 from mithro/ccsnoise-fix
    
    Fix ccsnoise properties ending up in basic liberty files.
```
steps done:
 1. git submodule update --init libraries/sky130_fd_sc_hs/latest
 2. make sky130_fd_sc_hs 
 3. I opened sky130_fd_sc_hs/latest/timing/sky130_fd_sc_hs__tt_025C_1v80.lib
 4. under sky130_fd_sc_hs__buf_1: timing(){ I find ccsn_first_stage ()

Is there anything I'm doing wrong in these steps?

mithro, 2020-07-14:
I can confirm that I am seeing this. It seems I removed the ccsnoise from pins but reverted the removal in the timing section.

mithro, 2020-07-14:
Actually, now I'm not seeing it...

msaligane, 2020-07-14:
I did a quick check in primetime of `sky130_fd_sc_hs`

I see sky130_fd_sc_hs__tt_025C_1v80.lib has these issues from primetime:
```
Error: Line 13353, Cell 'sky130_fd_sc_hs__a21bo_1', pin 'B1_N', The 'ccsn_dc' refers to a nonexistent or empty 'lu_table_template'. (LBDB-259)
Error: Line 13391, Cell 'sky130_fd_sc_hs__a21bo_1', pin 'B1_N', The 'ccsn_ovrf' refers to a nonexistent or empty 'lu_table_template'. (LBDB-259)
Error: Line 13397, Cell 'sky130_fd_sc_hs__a21bo_1', pin 'B1_N', The 'ccsn_ovrf' refers to a nonexistent or empty 'lu_table_template'. (LBDB-259)
Error: Line 13405, Cell 'sky130_fd_sc_hs__a21bo_1', pin 'B1_N', The 'ccsn_ovrf' refers to a nonexistent or empty 'lu_table_template'. (LBDB-259)
```

But other hs libs, like sky130_fd_sc_hs__tt_025C_1v89.lib and sky130_fd_sc_hs__tt_025C_2v10.lib, they have passed my tests with no errors.

msaligane, 2020-07-14:
For example if you look at a21bo cell in hs lib, there is no sky130_fd_sc_hs__a21bo_1__tt_025C_1v80.lib.json
see below:
```
sky130_fd_sc_hs__a21bo_1__tt_025C_1v20.lib.json
sky130_fd_sc_hs__a21bo_1__tt_025C_1v35.lib.json
sky130_fd_sc_hs__a21bo_1__tt_025C_1v44.lib.json
sky130_fd_sc_hs__a21bo_1__tt_025C_1v50.lib.json
sky130_fd_sc_hs__a21bo_1__tt_025C_1v62.lib.json
sky130_fd_sc_hs__a21bo_1__tt_025C_1v68.lib.json
**sky130_fd_sc_hs__a21bo_1__tt_025C_1v80_ccsnoise.lib.json**
sky130_fd_sc_hs__a21bo_1__tt_025C_1v89.lib.json
sky130_fd_sc_hs__a21bo_1__tt_025C_2v10.lib.json
sky130_fd_sc_hs__a21bo_1__tt_100C_1v80.lib.json
sky130_fd_sc_hs__a21bo_1__tt_150C_1v80.lib.json
```

mithro, 2020-07-15:
I have replicated the error and a fix should be in #102.

PR #92 (2020-07-13): Fix ccsnoise properties ending up in basic liberty files.
Fixes #77.
commit b626ca01 (2020-07-13): scripts/liberty: More aggressive ccsnoise filtering.

Fixes #77.
PR #102 (2020-07-15): scripts/liberty: Second attempt at filtering all ccsnoise data.
Fixes #77.
commit 463f2fd6 (2020-07-15): scripts/liberty: Second attempt at filtering all ccsnoise data.

Fixes #77.
closed 2020-07-15 by mithro