google/skywater-pdk #84: sky130_fd_sc_lp syntax issue in: comp_attribute power_gating_pin attribute

msaligane, 2020-07-10:
## Expected Behavior
correct format:
```
power_gating_pin("power_pin_1", \
			  "1");
```

## Actual Behavior
some cells in lp lib have the following attribute in their respective .json file:

`"comp_attribute power_gating_pin": "power_pin_1\", \"1”,
`
For example, this cell:
`sky130_fd_sc_lp/v0.0.1/cells/srsdfrtp/sky130_fd_sc_lp__srsdfrtp_1__ss_100C_1v60.lib.json
`
the \” sign will be converted to “ when generating the libs.

But the liberty.py has the following assertion line 838:
```

elif isinstance(l, str):
    print(l, (k, v))
    assert '"' not in l, (k, v)
    o.append('"%s"' % l)
which reports an error when “ is found.

```
This is a json file issue, I think you need to update the “power_gating_pin” attribute in json files in the lp lib.

mithro, 2020-07-11:
FYI - I have a fix that I'm pushing for this shortly.

mithro, 2020-07-11:
I have fixed the issue with `power_gating_pin` attribute in the `sky130_fd_sc_lp` library in https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_lp/+/1d2c6cfdb9794a722f4746481014307340fe0921%5E1..1d2c6cfdb9794a722f4746481014307340fe0921/ and integrated that change into the pull request https://github.com/google/skywater-pdk/pull/35

msaligane, 2020-07-11:
Great! Do you want me to test them still?

mithro, 2020-07-11:
FYI- https://github.com/google/skywater-pdk/issues/73 is a duplicate of this issue now I think?

msaligane, 2020-07-11:
Correct.

closed 2020-07-21 by mithro