google/skywater-pdk #85: Check the verilog for good style and run automated style / lint / formatting checks on it

mithro, 2020-07-10:
The verilog in this repository has been formatted to follow a consistent style but that style has not been documented anywhere. We should document the style and add automated checks which make sure the style stays consistent.

We may want to move to the [Verilog style of be that used by LowRISC projects](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md). That way we [could use the Verible tool](https://github.com/google/verible) for checking compliance and eventually auto-formatting.

We should also check that the automatic generated verilog output is consistent with the approved style too.

mithro, 2020-07-10:
FYI - @hzeller

mithro, 2020-07-10:
I believe this is blocked by issues like;
 - [ ] https://github.com/google/verible/issues/271
 - [ ] https://github.com/google/verible/issues/270
 - [ ] https://github.com/google/verible/issues/269

hzeller, 2020-07-11:
also `specify`...`endspecify`

Currently, verible formats about 22k out of the 31k verilog files.

I'll have to sieve through the other error messages; probably not too terrible - here are the unique ones:
```
     72 syntax error, rejected "$hold" (https://github.com/google/verible).
    297 syntax error, rejected "$recrem" (https://github.com/google/verible).
   1662 syntax error, rejected "$setuphold" (https://github.com/google/verible).
   1221 syntax error, rejected "$width" (https://github.com/google/verible).
   5838 syntax error, rejected "cell" (https://github.com/google/verible).
   2031 syntax error, rejected "endspecify" (https://github.com/google/verible).
   2256 syntax error, rejected "(" (https://github.com/google/verible).
  11463 syntax error, rejected "if" (https://github.com/google/verible).
```

mithro, 2020-07-11:
Could you group the errors by verilog file type? IE `.blackbox.v`, `.tb.v`, `.symbol.v`?

I'm guessing all the specify related errors come from the `XXX.specify.v` files?

hzeller, 2020-07-11:
Most of them actually happen in specify.v files, only the `cell` syntax error happens in *.v
 
```
     72 .specify.v  syntax error, rejected "$hold" (https://github.com/google/verible).
    297 .specify.v  syntax error, rejected "$recrem" (https://github.com/google/verible).
   1221 .specify.v  syntax error, rejected "$width" (https://github.com/google/verible).
   1662 .specify.v  syntax error, rejected "$setuphold" (https://github.com/google/verible).
   2031 .specify.v  syntax error, rejected "endspecify" (https://github.com/google/verible).
   2256 .specify.v  syntax error, rejected "(" (https://github.com/google/verible).
   5838 .v  syntax error, rejected "cell" (https://github.com/google/verible).
  11463 .specify.v  syntax error, rejected "if" (https://github.com/google/verible).
```

mithro, 2020-07-11:
Great!

So it sounds like if we fix the "rejected 'cell'" issue we can "auto-format" everything apart from the `XXX.specify.v` files?

hzeller, 2020-07-12:
Yep, looks like.

mithro, 2020-09-30:
Actually I think the 'cell' issue is caused by it being invalid Verilog -- see #152 ?

fangism, 2020-09-30:
It is a 2001 keyword:

```
/* The 1364-2001 configuration tokens. */
%token TK_cell "cell"
```