google/skywater-pdk #86: Write a "cell index / table" generator for the documentation

mithro, 2020-07-11:
It is very useful to have tables of the cells available in each library. This should be included in the Sphinx generated PDK documentation. The task here is to auto-generate these tables as either;
 - part of the Sphinx generation directly, or
 - As a standalone script in ([python-skywater-pdk package](https://github.com/google/skywater-pdk/tree/master/scripts/python-skywater-pdk)) which generates rst files committed to the repository ([like the generated verilog](https://github.com/google/skywater-pdk/pull/61) or image files).

This should be easily be done due to the library structure being based around each cell having its own directory and a `definition.json` file which describes the basic shape and features of the cell.

Here is an example from a spreadsheet I created;
-----
[image]
[image]
-----

Here is an example [from the libresilicon Standard Cell Library Build Report](https://pdk.libresilicon.com/dist/StdCellLib_20200624/Catalog/buildreport.html);
-----
[image]
-----

Here is an example from [this random document](http://web.engr.oregonstate.edu/~traylor/ece474/reading/SAED_Cell_Lib_Rev1_4_20_1.pdf)
----
[image]
----

Here is another example from [a random thesis on creating standard cells](https://shareok.org/bitstream/handle/11244/10180/Ahmed_okstate_0664M_10739.pdf?sequence=1)
----
[image]
----

mithro, 2020-08-17:
Example cell - https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/a2111o/ 

Definition file which contains a summary of the cell info - https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/a2111o/definition.json

PiotrZierhoffer, 2020-08-25:
I used some of the `definitions.json` data to create a basic list. The script just takes a path to the library and creates a file. I also modified the README.rst manually to include the file inline (so we don't have to parse it when regenerating).

How do I submit the results for review? Create a sample PR in one of the submodules maybe?

PiotrZierhoffer, 2020-08-25:
Also it now takes data both from `cells` and `models`. This can obviously be changed if needed, but seemed to make sense

mithro, 2020-10-22:
The issue at https://github.com/google/skywater-pdk/pull/143 does some initial work here.

proppy, 2022-05-30:
@kgugala @PiotrZierhoffer found #143 #256 and #266 that all seem to contribute to fixing this, could you advise on the best way to merge them (rebase, apply in order, drop duplicate changes?). Are there other changes from https://github.com/antmicro/skywater-pdk/ that need to be added as well?

wgryncewicz, 2022-06-08:
@proppy I'll take a look into this.

PR #266 (2023-05-10): 86 cell cross index
This PR adds the generator of a cell index, in a form of cross-library `cell name` / `lib name` table.
The table is based on spreadsheet view, quoted in the issue description.
Each entry links to corresponding cell description page.
[Generated index view](https://antmicro-skywater-pdk-docs.readthedocs.io/en/86-cell_cross_index/contents/cell-index.html)

Fixes #86 

_Note 1_: The PR includes cell readme generator, which was already submitted as part of [#256](https://github.com/google/skywater-pdk/pull/256) . This version add Sphinx ext wrapping and pre-build event trigger.
 [... cut]
closed 2023-05-10 by mithro