google/skywater-pdk #87: Actually run the test bench, record values into output VCD file and render them with wavedrom in documentation

mithro, 2020-07-11:
The PDK currently has auto-generated "test" benches for all the standard cells and models. See the following examples;
 - https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/a2111o/sky130_fd_sc_hd__a2111o.tb.v
 - https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/sdfbbn/sky130_fd_sc_hd__sdfbbn.tb.v

It would be good to take these test benches, run them through the simulator and saved the output to VCD files. Any open source simulator like Verilator or Icarus Verilog could be used for this.

Once there are VCD files, these should then be rendered with wavedrom to give [nice waveform diagrams](https://wavedrom.com/editor.html?%7B%20signal%3A%20%5B%0A%20%20%7B%20name%3A%20%27A%27%2C%20wave%3A%20%2701..0..%27%2C%20%20node%3A%20%27.a..e..%27%20%7D%2C%0A%20%20%7B%20name%3A%20%27B%27%2C%20wave%3A%20%270.1..0.%27%2C%20%20node%3A%20%27..b..d.%27%2C%20phase%3A0.5%20%7D%2C%0A%20%20%7B%20name%3A%20%27C%27%2C%20wave%3A%20%270..1..0%27%2C%20%20node%3A%20%27...c..f%27%20%7D%2C%0A%20%20%7B%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20node%3A%20%27...g..h%27%20%7D%0A%5D%2C%0A%20%20edge%3A%20%5B%27b-%7Ca%20t1%27%2C%20%27a-%7Cc%20t2%27%2C%20%27b-%7C-c%20t3%27%2C%20%27c-%7C-%3Ee%20t4%27%2C%20%27e-%7C%3Ef%20more%20text%27%2C%20%27e%7C-%3Ed%20t6%27%2C%20%27c-g%27%2C%20%27f-h%27%2C%20%27g%3C-%3Eh%203%20ms%27%5D%0A%7D%0A) in [the documentation on ReadTheDocs](https://skywater-pdk.rtfd.io). This could use be done with one of the wavedrom Sphinx plugins.

kkiningh, 2020-07-15:
I don't think you can use Verilator for this without changing the test benches since Verilator doesn't support simulating delays (e.g. [line 110](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/a2111o/sky130_fd_sc_hd__a2111o.tb.v#110) here).

mithro, 2020-07-15:
@kkiningh - If I recall correctly, support as added recently?

kkiningh, 2020-07-15:
In the [FAQ](https://www.veripool.org/projects/verilator/wiki/Manual-verilator#FAQ-FREQUENTLY-ASKED-QUESTIONS) it says "All delays (#) are ignored, as they are in synthesis." Maybe the documentation hasn't been updated?

mithro, 2020-07-15:
If I'm correct, it would have probably only been in the last few months as Verilator pushes forward supporting more SystemVerilog and UVM stuff.

kkiningh, 2020-07-15:
I get the following error running with Verilator HEAD.

```
%Warning-STMTDLY: test.v:29:10: Unsupported: Ignoring delay on this delayed statement.
                              : ... In instance top
   29 |         #20   A1   = 1'b0;
      |          ^~
                  ... Use "/* verilator lint_off STMTDLY */" and lint_on around source to disable this message.
```

mithro, 2020-07-15:
Guess I'm wrong. Let's start with Icarus first then?