google/skywater-pdk #91: Publish GDS layer map

Laksen, 2020-07-12:
Currently there doesn't seem to be any way to find the GDS layer id for a given layer.

RTimothyEdwards, 2020-07-12:
@Laksen :  This document: https://docs.google.com/spreadsheets/d/176OcP2h_TT_pDP2-zBLijlgFUP38LVgMQPlDtpLcrsk/edit?ts=5eff8387#gid=459375686 doesn't tell you what you want to know?

smunaut, 2020-07-12:
Also see https://github.com/google/skywater-pdk/pull/89 that aims to merge that document inside this repo.

Laksen, 2020-07-12:
@RTimothyEdwards yes, thank you, that contains the relevant information. And with the PR @smunaut linked it should be easy to generate a layer map from it

mithro, 2020-07-19:
Adding the resistance and capacitance values in https://github.com/google/skywater-pdk/pull/106

RTimothyEdwards, 2020-07-22:
@mithro :  Is there anything that actually needs to be done here for this issue?

20Mhz, 2020-07-28:
maybe ASCII map files?

rovinski, 2020-07-28:
@20Mhz there isn't really a standard for ASCII map files, and every tool seems to have a different format for them.

#89 added a CSV file version of the Google doc. IMO if there are any other versions of a GDS layer map, they should be extracted from that CSV via script to maintain a singular layer map source file.