google/skywater-pdk #95: Mismatch between .lef and .lib files around bias pins (VNB and VPB)

mithro, 2020-07-13:
@msaligane -- Can you add the description here?

msaligane, 2020-07-13:
## Expected Behavior
No ports mismatch between LEF and LIB files.
   - Remove VNB and VPB from the liberty files will fix this issue.
   - Buffer/delay/inverter footprints are recognized.

## Actual Behavior

innovus/15.21.000 can only detect buffers and inverters when there is an exact match between LEF and lib files (IE if vpb/vnb are deleted from the .lib it works)
innovus/18.10.000 can detect buffers and inverters if vpb/vnb are defined in lower case or deleted.. but fails to detect them if they are defined as upper case.
Innovus uses its "footprintless" flow except if we provide a footprint file (unusual). And if there are mismatches between .libs and .lefs then innovus fails to recognize delay/buffer/inverter cells.

msaligane, 2020-07-13:
I will close the issue I logged.

mithro, 2020-07-13:
@msaligane Can you list the cells where they don't match?

msaligane, 2020-07-13:
@mithro all of them actually.

20Mhz, 2020-07-13:
how about adding libertarian attribute physical_connection: device_layer? would that help? Semantic seems to imply the actual connection is outside

rovinski, 2020-07-13:
I agree with @20Mhz. The right solution looks to be properly annotating the connections. Here is a sample of what this might look like:
```
pg_pin (VNB) {
  pg_type : nwell;
  physical_connection : device_layer;
  voltage_name : "VNB";
}
pg_pin (VPB) {
  pg_type : pwell;
  physical_connection : device_layer;
  voltage_name : "VPB";
}
pg_pin (VPWR) {
  pg_type : primary_power;
  related_bias_pin : "VNB";
  voltage_name : "VPWR";
}
pg_pin (VGND) {
  pg_type : primary_ground;
  related_bias_pin : "VPB";
  voltage_name : "VGND";
}
```
These pins should also be defined in the LEF on the nwell / pwell layers, if they aren't already.

mithro, 2020-07-13:
I'm pretty sure we already have those?

msaligane, 2020-07-13:
@mithro VPB/VNB aren't there in the lef files. Adn I do not think they can be defined. The connection to the wells is that through tap cells.

I have no preferences of deleting these pins from the .libs or adding in the lef. 
@rovinski there is no pin for nwell/pwell in any standard cell (the tap cell takes care of that)...
AFAIK, you only find that in fdsoi or ddc technologies.

mithro, 2020-07-13:
Looking at https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/refs/heads/master/cells/a2111o/sky130_fd_sc_hd__a2111o_1__ff_100C_1v65.lib.json#137
It seems like we should clean up the definition to match what @rovinski suggested.

We can also add an output mode for the liberty generator to skip the bias pins.

I'm going to split this into two seperate GitHub issues.

msaligane, 2020-07-13:
@mithro Sounds good. Let me know when it is available to test. The .lib modification should probably fix this issue (without changing the LEF files).

rovinski, 2020-07-13:
@msaligane other commercial PDKs have pwells and nwells exposed as pins in the LEF. Having a LEF pin exposed does not imply anything about connectivity from the well to other parts of the cell. Tapcells have internal connections from the rails to the wells, but that doesn't mean that those are the only cells which should have well pins. I'm pretty sure if one wants to use any kind of biasing strategy, you need those pins exposed.

mithro, 2020-07-13:
I have split this into #97 and #98 and hence closing this issue.

closed 2020-07-13 by mithro