google/skywater-pdk #96: Standard cell LEF views disagree on whether LI connected to power rail is a pin or an obstruction

RTimothyEdwards, 2020-07-13:
@mithro :  The scripts that we used to ensure that the LI and MCON layers under the power rails would show up in all of the standard cell LEF files still has one problem---some cells are now adding the LI and MCON under the power rails to the geometry of the power rail pins, while other cells are showing those layers as obstructions.  The latter may make somewhat more sense logically (connect only to the MET1 part of the power rail because it has the lowest resistance).

One way to resolve the issue might be for me to add an option to magic's "lef write" command to insist that pins should only be created on the top-most layer for a pin, and then all layers under that would be treated as obstructions.  I think that would produce a result that would make all cells match.

I will work on this modification and let you know when it is done so that you can add the "lef write" option to the scripts and re-run to correct the LEF views.

mithro, 2020-07-15:
@RTimothyEdwards Does this affect all the standard cells or only specific libraries?

RTimothyEdwards, 2020-07-15:
@mithro :  It is effectively the difference between those LEF views that only exist from magic because SkyWater didn't provide them (antenna diode and decap cells, mostly), and the original SkyWater LEF views.  But I just updated magic (have not pushed yet) with an option "lef write -toplayer" that produces the same output as the SkyWater LEF, which is to output only the topmost layer that a pin is connected to as part of the PIN record geometry, and place all the layers underneath in the OBS geometry.  Updating magic and updating the scripts that generate the .magic.lef views and running it all again should fix the problem.  I am currently trying to figure out why magic does not output anything on MASTERSLICE layers (relevant to the other discussion on the VNB/VPB pins);  as soon as I have fixed that I will commit to the magic repo.

RTimothyEdwards, 2020-07-15:
@mithro :  I have committed/pushed the change to magic for the "lef write -toplayer".  If you update magic and use that command option in the script that generates the .magic.lef views, they will match the way that SkyWater shows the power rails as having pin geometry only on metal1 and the mcon and li1 layers being designated as obstruction layers.

mithro, 2020-07-20:
@RTimothyEdwards Will push an updated standard cells shortly.

PR #116 (2020-07-21): Updating `.magic.lef` files using `lef write {cell} -toplayer`
Fixes #96 - "Standard cell LEF views disagree on whether LI connected to power rail is a pin or an obstruction"

 - Updating [`sky130_fd_sc_hd` latest](./libraries/sky130_fd_sc_hd/latest) to v0.0.2-14-ga1fffcda
 - Updating [`sky130_fd_sc_hd` v0.0.1](./libraries/sky130_fd_sc_hd/v0.0.1) to v0.0.1-7-g76c4a7c4
 - Updating [`sky130_fd_sc_hd` v0.0.2](./libraries/sky130_fd_sc_hd/v0.0.2) to v0.0.2-14-ga1fffcda
 - Updating [`sky130_fd_sc_hdll` latest](./libraries/sky130_fd_sc_hdll/latest) to v0.1.1-16-g465b12fa
 - Updating [`sky130_fd_sc_hd [... cut]
closed 2020-07-21 by mithro