google/skywater-pdk #97: Improve the pg_pin definitions in the liberty timing files

mithro, 2020-07-13:
Currently the pg_pin definitions look like this;
```json
"pg_pin VGND": {
 "pg_type": "primary_ground",
 "voltage_name": "VGND"
 },
 "pg_pin VNB": {
 "pg_type": "primary_ground",
 "voltage_name": "VNB"
 },
 "pg_pin VPB": {
 "pg_type": "primary_power",
 "voltage_name": "VPB"
 },
 "pg_pin VPWR": {
 "pg_type": "primary_power",
 "voltage_name": "VPWR"
 },
```

From https://github.com/google/skywater-pdk/issues/95#issuecomment-657771298 we see they should look like this;
> I agree with @20Mhz. The right solution looks to be properly annotating the connections. Here is a sample of what this might look like:
> ```
> pg_pin (VNB) {
>   pg_type : nwell;
>   physical_connection : device_layer;
>   voltage_name : "VNB";
> }
> pg_pin (VPB) {
>   pg_type : pwell;
>   physical_connection : device_layer;
>   voltage_name : "VPB";
> }
> pg_pin (VPWR) {
>   pg_type : primary_power;
>   related_bias_pin : "VNB";
>   voltage_name : "VPWR";
> }
> pg_pin (VGND) {
>   pg_type : primary_ground;
>   related_bias_pin : "VPB";
>   voltage_name : "VGND";
> }
> ```
> These pins should also be defined in the LEF on the nwell / pwell layers, if they aren't already.

msaligane, 2020-07-13:
@mithro You can make a fake pin for VNB/VPB for each cell in the LEF. But there is no actual/physical pin there. So it doesn't really make sense to me. 
@RTimothyEdwards can you comment on that?

mithro, 2020-07-13:
@20MHz and @rovinski -- What would be the value for `device_layer`?

20Mhz, 2020-07-13:
device_layer is the value, meaning the connection is made outside, vs routing_pin, which means router needs to reach, routing pin may be needed for insulated wells if these are not tied to power rails

mithro, 2020-07-13:
Oh -- not "a device_layer" -- the "value 'device_layer'".

rovinski, 2020-07-13:
@mithro correct, it's literally the value 'device_layer'
@msaligane The VNB pin can use the nwell layers as the PORT shape.
```
PIN VNB
  DIRECTION INOUT ;
  USE GROUND ;
  PORT
    LAYER [nwell layer name] ;
      RECT [insert rect coordinates] ;
  END
END VNB
```
If this technology doesn't have a layer for defining pwell within the cell shape, then it would have to be a pin with no shape.

```
PIN VPB
  DIRECTION INOUT ;
  USE GROUND ;
END VPB
```

I'm not sure if this would cause LVS issues, though.

msaligane, 2020-07-13:
Thanks @rovinski . I guess we would have to try it :-P

RTimothyEdwards, 2020-07-14:
@msaligane , @rovinski , @mithro :  I ran across this problem yesterday, attempting to run LVS from a layout in qflow.  I realized that the original SkyWater sources DO have the VNB and VPB pins in the LEF file, but because they didn't have layers defined to attach them to, they put them on MET1, which makes no sense at all----that forces them to be connected to power and ground (which they aren't);  in Magic, it loses the extra pins when it extracts the file because it sees them as the same node.

I think the correct solution might be to define the well and substrate layers in the technology LEF file as type "MASTERSLICE", which will make them valid layers but not valid routing layers.  That would mean that the pins are valid across all views, including LEF, liberty, CDL, and GDS.  I will take a stab at rewriting one library this way and see if it solves the issues like I think it will (that will have to be confirmed on commercial tools;  I can only vouch for Magic and Netgen).

rovinski, 2020-07-14:
@RTimothyEdwards yes, that seems like the right thing to do. I see that nwell and pwell typically defined as `MASTERSLICE` layers in the tech lef.

It would be easy to define a shape for the nwell as there is a drawing layer in the GDS. But I'm not sure what to do about the shape for pwell because there is no pwell drawing layer. Maybe a shape isn't required if it's a `MASTERSLICE` layer.

RTimothyEdwards, 2020-07-14:
@rovinski @msaligane @mithro :  Provisionally, this solution is working for me.  I wrote a simple python script to parse the LEF file and search for all VBP and VNB pins, and change them from the incorrect "met1" or "li1" layer to "nwell" and "substrate", respectively.  Then I added "MASTERSLICE" entries in the technology file for "nwell" and "substrate".  Finally, I added the masterslice entries to the magic techfile, then ran the whole thing through installation.  The result was that I could use the abstract views in the layout and get it to pass LVS properly.

@rovinski :  As far as I can tell, the "masterslice" layers don't have to correspond to anything, but it remains to be seen whether the place and route tools, especially where the tap cells are added, see it as valid.  However, as you mentioned, this looks to me like a typical use case of "masterslice", and what "masterslice" was intended to be used for.

The simple python script I used is this:

[vpb_vnb_convert.py.zip](https://github.com/google/skywater-pdk/files/4921090/vpb_vnb_convert.py.zip)

RTimothyEdwards, 2020-07-14:
In reference to the above, and earlier comments in this thread, "pwell" is a better name than "substrate" and should be used instead.

msaligane, 2020-07-14:
@RTimothyEdwards @rovinski This is great. We still need to run calibre to cross check this fix but thanks for looking at LVS as well

RTimothyEdwards, 2020-07-14:
@msaligane :  Reading further into the LEF/DEF spec, there is some possibility that this is only valid for LEF version 5.7+.  The definition of MASTERSLICE itself says that the router can only use one MASTERSLICE and that if a pin appears on a MASTERSLICE layer, then a via must be defined between the MASTERSLICE and first routing layer. . . none of that would be consistent with nwell or pwell.  But, the 5.7 spec says that MASTERSLICE can have the following inside the layer definition:

   PROPERTY LEF58_TYPE "TYPE NWELL ;" ;
or
   PROPERTY LEF58_TYPE "TYPE PWELL ;" ;

which seems pretty definitive that it is meant to be used for nwell and pwell connections.

RTimothyEdwards, 2020-07-14:
@msaligane :  I'm not sure how or if this affects Calibre.  My own particular problem with LVS is that I have been using abstract (LEF) views in Magic so that I can extract a netlist that does not go down to the transistor layer but treats all circuits with LEF views as "black box" entries for the purpose of LVS.  For the hierarchical LVS to work, then, it's necessary that the LEF views have all the power pins represented, especially as the tap cells connect them together.  I still have some experiments to do with magic's generation of LEF from the GDS + pin annotation from the SkyWater LEF. . . after correcting for the layers of VNB and VPB.

rovinski, 2020-07-14:
Agreed, I don't think this should affect GDS-based LVS like Calibre. The only circumstance I can think of is if the schematic netlist contains the bulk-to-well connections but it doesn't extract those those connections from the layout. Not sure what the solution would be there; I know Calibre has miscellaneous bulk connection extraction options. I think we cross that bridge when we get there.

msaligane, 2020-07-14:
@RTimothyEdwards I am pushing through the temperature sensor design so I can test this. But I guess if there is a strive design, I can try to run LVS on it. My experience is that Calibre needs to understand all the power connection correctly. And since we are explicitly uncovering these 2 pins in LEF/.libs and thus verilog/cdl then it should be easier to debug. Now we will have issues for sure to get that running the first time.

RTimothyEdwards, 2020-07-15:
@mithro :  I have committed a change to magic to fix an issue that was preventing it from generating any output on "masterslice" layers when doing "lef write".  Now with a change in the LEF definitions in the magic techfile to add nwell and pwell as "masterslice" layers, it generates output with VNB attached to pwell (substrate) and VPB attached to nwell, where the whole nwell layer is also represented in the LEF view generated by magic (this should resolve all my own issues running magic and netgen).  The technology LEF files need to be updated with the "masterslice" entries for nwell and pwell, also.  If we agree that this is the way it should be implemented, then all these things should be done at once (updating magic, updating the magic techfile, updating the technology LEF files, and regenerating the .magic.lef views of all the standard cells) and tested.

mithro, 2020-07-20:
I will push an updated standard cells soon.

mithro, 2020-07-21:
With https://github.com/google/skywater-pdk/pull/112 (Fix power pins) merged and https://github.com/google/skywater-pdk/pull/116 (Updating `.magic.lef` files using `lef write {cell} -toplayer`) merged, I believe this is fixed now.

closed 2020-07-21 by mithro