google/skywater-pdk #98: Add support for removing the bias pins in output timing files

mithro, 2020-07-13:
The bias pins in the liberty timing files cause issues with some versions of Innovus.

Just like we have the ability to create ccsnoise and leakage versions of the liberty files, we can make the script only output the bias pins if requested.

mithro, 2020-07-13:
See https://github.com/google/skywater-pdk/issues/95#issuecomment-657696690
> ## Expected Behavior
> No ports mismatch between LEF and LIB files.
> 
> * Remove VNB and VPB from the liberty files will fix this issue.
> * Buffer/delay/inverter footprints are recognized.
> 
> ## Actual Behavior
> innovus/15.21.000 can only detect buffers and inverters when there is an exact match between LEF and lib files (IE if vpb/vnb are deleted from the .lib it works)
> innovus/18.10.000 can detect buffers and inverters if vpb/vnb are defined in lower case or deleted.. but fails to detect them if they are defined as upper case.
> Innovus uses its "footprintless" flow except if we provide a footprint file (unusual). And if there are mismatches between .libs and .lefs then innovus fails to recognize delay/buffer/inverter cells.

RTimothyEdwards, 2020-07-22:
@mithro :  This issue is related to #114 and probably should be merged, although "adding support for removing the bias pins" is maybe independent of "removing the bias pins"?  Anyway, my answer to #114 is that this issue is going about it the wrong way; i.e., saying that the bias pins should be removed from the .lib files whereas I believe that the correct solution is to add them to the LEF files by way of defining the nwell and pwell layers in the technology LEF files.  See my comments in #114 for a more detailed explanation.

egiacomin, 2021-03-27:
@mithro @RTimothyEdwards this issue is quite old, but thought I would chime in real quick. I also agree that the right way is to add the pins in the .lef file rather than removing them from the .lib files.
Looks like it has been done in the latest version of the librairies. I managed to do a run on Innovus 20.13 and everything worked great. Should this issue be closed?

RTimothyEdwards, 2021-03-27:
@egiacomin :  I agree that since the bias pins were added to the LEF files, and this has been largely verified (except for some recent report about required CUT and OVERLAP statements that may also be needed by some tools), this issue probably should be closed.

egiacomin, 2021-03-27:
@RTimothyEdwards Indeed they seem to work fine now. By the way, I'm the one who mentioned the CUT and OVERLAP issues, but these ones are related to the "technology lef" file rather than the "standard cell" lef file mentioned here.