google/skywater-pdk #99: Rework the .lib generation in the Makefile

mithro, 2020-07-14:
There should be Makefile targets for each `.lib` that needs to be generated.

For example - `make libraries/sky130_fd_sc_ms/v0.0.2/timing/sky130_fd_sc_ms__ff_n40C_1v95.lib` should work.

Currently;
```
make[1]: Entering directory '/home/tim/github/google/skywater-pdk'
make[1]: *** No rule to make target 'libraries/sky130_fd_sc_ms/v0.0.2/timing/sky130_fd_sc_ms__ff_n40C_1v95.lib'.  Stop.
make[1]: Leaving directory '/home/tim/github/google/skywater-pdk'
```

mithro, 2020-07-14:
It is unclear to me if we need the Python scripts to generate a `make depend` like output to make this happen.

smunaut, 2020-07-14:
You can probably get away with just making it depend on all the individual cells .lib files

mithro, 2020-07-14:
Any idea how to handle the `--ccsnoise` and `--leakage` versions?

rovinski, 2020-07-14:
Could use the following rules:
```
CCSNOISE_LIBS = $(wildcard libraries/*/*/timing/*_ccsnoise.lib.json)
PWRLKG_LIBS = $(wildcard libraries/*/*/timing/*_pwrlkg.lib.json)
STANDARD_LIBS = $(filter-out $(CCSNOISE_LIBS) $(PWRLKG_LIBS),$(wildcard libraries/*/*/timing/*.lib.json))

$(CCSNOISE_LIBS):
    command_for_ccsnoise
$(PWRLKG_LIBS):
    command_for_pwrlkg
$(STANDARD_LIBS)
    command_for_standard
```

smunaut, 2020-07-14:
```
LIBRARIES=$(basename $(filter-out %__common.lib.json,$(wildcard libraries/*/*/timing/*.lib.json)))

libraries/%_ccsnoise.lib: libraries/%_ccsnoise.lib.json
        python -m skywater_pdk.liberty \
                $(shell echo $@ | sed -e's/^.*\/\(.*\)\/.*\/\(.*\)__\(.*\)_ccsnoise\.lib$$/libraries\/\2\/\1/' ) \
                $(shell echo $@ | sed -e's/^.*__\(.*\)_ccsnoise\.lib$$/\1/') \
                --ccsnoise

libraries/%_pwrlkg.lib: libraries/%_pwrlkg.lib.json
        python -m skywater_pdk.liberty \
                $(shell echo $@ | sed -e's/^.*\/\(.*\)\/.*\/\(.*\)__\(.*\)_pwrlkg\.lib$$/libraries\/\2\/\1/' ) \
                $(shell echo $@ | sed -e's/^.*__\(.*\)_pwrlkg\.lib$$/\1/') \
                --leakage

libraries/%.lib: libraries/%.lib.json
        python -m skywater_pdk.liberty \
                $(shell echo $@ | sed -e's/^.*\/\(.*\)\/.*\/\(.*\)__\(.*\)\.lib$$/libraries\/\2\/\1/' ) \
                $(shell echo $@ | sed -e's/^.*__\(.*\)\.lib$$/\1/')

all: $(LIBRARIES)

clean:
        rm -f $(LIBRARIES)
```

This is my best attempt so far.

It properly :
 * List all checked out libraries and skipped un-init submodules
 * Generate them for all process corners and variants (ccsnoise / leakage)
 * Supports parallel builds ( -j N )
 * Supports make clean
 * Support individual build ( `make libraries/sky130_fd_sc_ms/v0.0.2/timing/sky130_fd_sc_ms__ff_n40C_1v95.lib` )

What it doesn't support is the dependencies / rebuild if any of the cell .json changes or if a new cell is added or that kind of stuff.

Also I just noticed that if there is a ccsnoise version you need to generate the "standard" version from the ccsnoise `.json` and it doesn't do that ATM.

mithro, 2020-07-14:
@smunaut - Do you want to send that as a pull request?

mithro, 2020-07-14:
@smunaut - We could also make the `python -m skywater_pdk.liberty` tool take the Make argument rather than running it through sed?

mithro, 2020-07-14:
We could also make the `python -m skywater_pdk.liberty` tool have a mode where it generates a Make dependency file (like the C compiler does for headers)?

rovinski, 2020-07-15:
@smunaut please don't use sed shell callouts... there are much cleaner ways to do that via makefile functions. Give me a little time and I can get something cleaner.

mithro, 2020-07-15:
@rovinski -- I *think* the best solution is to just make the python script accept the output file as an argument?

rovinski, 2020-07-15:
@mithro if the python script could parse the details it needs from the filename, then yes, that would be even cleaner.

mithro, 2020-07-20:
Does someone here want to send a pull request with the fixes described here?

smunaut, 2020-07-20:
I can take a stab at it ( adding a 'make' mode to the script so parsing and dependencies is done internally ) , but this might not be before the week end.

smunaut, 2020-07-22:
Ok, I started looking at it, doing most of the work in the python script.
Seems to work so far but I still need to add the dependency tracking.