google/skywater-pdk #104: Missing "END LIBRARY" Statement in Cell LEF files.

acasper45, 2020-07-19:
The current LEF files generated for the Skywater PDK cells lack an "END LIBRARY" statement at the end of the file. Without this statement the OpenRoad LEF parser complains with an error message 

```Error: ERROR (LEFPARS-1002): Incomplete lef file.``` 

Adding "END LIBRARY" to the last line of the LEF file eliminates this error and the cells are imported without issue.

Unless the "END LIBRARY" was purposefully removed seems like it would be useful to add, even if we do ultimately end up merging all the LEFs into a single file (e.g. [issue 52](https://github.com/google/skywater-pdk/issues/51).)

mithro, 2020-07-19:
@acasper45 Does there need to be some type of "start library"?

acasper45, 2020-07-19:
I was wondering the same thing but I don't think there is need for a "start library" type statement. I'm no LEF file expert by any means but looking at existing examples and [this reference](http://www.ispd.cc/contests/18/lefdefref.pdf) it seems an "END LIBRARY" is all thats required.

RTimothyEdwards, 2020-07-20:
@mithro :  This is probably caused by Magic, which does not put "END LIBRARY" at the end of single macro output files.  Note that this is because _there is no documented use of the statement "END LIBRARY" in the LEF/DEF spec_!  Therefore I maintain that the commercial tools are in error.  However, since according to the above, it appears that "END LIBRARY" is what commercial tools want to see at the end of every LEF file (in spite of the complete uselessness of such a statement), I have added it to the output that Magic generates for single macros.

RTimothyEdwards, 2020-07-20:
Also:  No, there is no "LIBRARY" statement corresponding to "END LIBRARY", although that is the way "END" is used in _every single other LEF/DEF "END" statement_.  Once again, LEF/DEF spec fails to make internally consistent sense.

mithro, 2020-07-20:
I will push standard cells with this updated.

mithro, 2020-07-21:
This is now fixed!

PR #117 (2020-07-21): Making sure all .lef files end in `END LIBRARY`
Fixes #104.

 - Updating [`sky130_fd_sc_hd` latest](./libraries/sky130_fd_sc_hd/latest) to v0.0.2-16-gd48b3285
 - Updating [`sky130_fd_sc_hd` v0.0.1](./libraries/sky130_fd_sc_hd/v0.0.1) to v0.0.1-8-g396e3a55
 - Updating [`sky130_fd_sc_hd` v0.0.2](./libraries/sky130_fd_sc_hd/v0.0.2) to v0.0.2-16-gd48b3285
 - Updating [`sky130_fd_sc_hdll` latest](./libraries/sky130_fd_sc_hdll/latest) to v0.1.1-18-gf98e27af
 - Updating [`sky130_fd_sc_hdll` v0.1.0](./libraries/sky130_fd_sc_hdll/v0.1.0) to v0.1.0-9-g74c3af61
 - Updating [`sky130_fd_sc_hdll` v0.1.1](./ [... cut]
closed 2020-07-21 by mithro