google/skywater-pdk #105: All periphery rules should include units

mithro, 2020-07-19:
All the periphery rules in the documentation should include units.

pkatarzynski, 2020-11-05:
Judging from the CSV files located here https://github.com/google/skywater-pdk/tree/master/docs/rules/periphery a numeric value associated with a certain rule has one unit ( [um] I presume) unless specified differently in the rule description. There is no coherent formating for the units - they are expressed like mm [mm] (mm). 
I think we can parse the rule text, look for those units and assign them to the associated value as long as there is only one unit-like string detected.

pkatarzynski, 2020-11-05:
Well there are rules which do not follow this pattern. Is there a chance we can introduce a "unit" column in the source CSV files?
We can populate it with a script which will do the unit searching as described above. Some of the missing units will need to be added manually but it looks like a single-pass process.

pkatarzynski, 2020-11-05:
@mithro

mithro, 2020-11-05:
@pkatarzynski Introducing a "unit" column in the source CSV file seems like a good idea.

mithro, 2020-11-05:
We should also clean up how they are expressed.

PR #227 (2020-11-17): Add units to rules
Added a function which generates 'unit' column when parsing `periphery.csv` to `periphery-rules.rst`.
The unit is detected from description (if stated explicitly) or assumed basing on description keywords.
The script assumed 1 um as a default rule length unit.
Fixes #105 

Tests:
https://antmicro-skywater-pdk-docs.readthedocs.io/en/105_add_units_to_rules/rules/periphery.html
closed 2020-11-17 by mithro