google/skywater-pdk #107: Missing subcells "../u_vpwr_vgnd/sky130_fd_sc_hs__u_vpwr_vgnd.v" in inv module

Gaurav-kgk, 2020-07-20:
## Expected Behavior
The behavioral sub-module of inverter module imports subcell **include "../u_vpwr_vgnd/sky130_fd_sc_hs__u_vpwr_vgnd.v"**, but the file is missing in the PDK library and hence parse error is coming on compiling

## Actual Behavior
The subcell that is being imported should be present in order to compile the verilog file of inverter.

## Steps to Reproduce the Problem

1. Include the missing file so that the parse error is solved
1.
1.

## Specifications

- Version: 
- Platform:

RTimothyEdwards, 2023-07-06:
@jalcim 

The relevant file is called `sky130_fd_sc_hs__udp_pwrgood_pp_pg.v` (or I should say that's the file that appears to have been derived from the original `u_vpwr_vgnd.v`).  But it names the primitive `sky130_fd_sc_hs__udp_pwrgood_pp$PG`.  This name translation appears to have been missed in all the files.

In all cases the include file should be `.include ../../models/udp_pwrgood_pp_pg/sky130_fd_sc_hs__udp_pwrgood_pp_pg.v`, and the name of the UDP primitive in all the functional verilog views should be changed from `u_vpwr_vgnd0` to `sky130_fd_sc_hs__udp_pwrgood_pp$PG`.

RTimothyEdwards, 2023-07-07:
[correct_hs_vlog.py.gz](https://github.com/google/skywater-pdk/files/11973526/correct_hs_vlog.py.gz)
@jalcim :  Copy this script to the cells/ directory and run it.  It will patch the verilog files and correct the issue.

zznjupt, 2025-05-09:
> [correct_hs_vlog.py.gz](https://github.com/google/skywater-pdk/files/11973526/correct_hs_vlog.py.gz) [@jalcim](https://github.com/jalcim) : Copy this script to the cells/ directory and run it. It will patch the verilog files and correct the issue.

It seems that the script does not resolve all path errors, such as in some cell files: "../u_df_p_no_pg/sky130_fd_sc_hs__u_df_p_no_pg.v"

RTimothyEdwards, 2025-05-09:
There are a lot of errors in the HS library that need to be sorted through and corrected.  The include paths is only one of them.  There are also cases in which the included file has been named something else and does not exist, and the cell name of the UDP cell is wrong.  I would like to sort out these issues and get the library corrected.

Please note that the location of the libraries has changed.  Please direct all future issues to https://github.com/fossi-foundation/skywater-pdk-libs-sky130_fd_sc_hs/issues.

I will create a new issue there to cover these issues.