google/skywater-pdk #114: Issue with vpb/vnb rails layers can't be selected in virtuoso

msaligane, 2020-07-21:
## Expected Behavior
I am expecting to be able to select these layers or draw them in virtuoso

## Actual Behavior
I imported `sky130_fd_sc_hs__nor2_1` to inspect a reference layout of a standard cell. However, I am not able to draw these 2 layers (highlighted in red) in my own standard cells. 
[image]

mithro, 2020-07-21:
@msaligane - I'm unsure what is causing this nor how to fix it, any ideas?

msaligane, 2020-07-21:
@mithro I am not sure why this is happening. Never see this happening before.
From what I understood, these are related to nsdm/psdm layers which surround ndif or pdif shapes, including the n+ and p+ shapes needed to make well-taps. Magic auto-generated them when it generated gds's from .mag.
@RTimothyEdwards Could you please help here? 
Meanwhile I will import a cell from the original gds to see if I have the same issue.

msaligane, 2020-07-21:
This is what I see in the original libraries:
[image]

mithro, 2020-07-21:
@msaligane - There should be no difference in the GDS between the original cell and the new cells (except a few strings). We should figure out what is going wrong here....

msaligane, 2020-07-22:
@mithro I am trying to understand as well. Both the old libraries and new ones seems to pass LVS on this cell. So I do not think the difference on this layer causes a problem. It actually seems more like a tag since I can't select the layers highlighted above.

msaligane, 2020-07-22:
@mithro is there a layermap available. These are some warnings I see when I import the scs8hd lib:

```
WARNING (XSTRM-36): The objects in the stream layer-datatype '92:0' are ignored. This is because the stream layer-datatype pair is not specified in the layer map file. Specify the stream layer-datatype pair in the layer map file and try again.
WARNING (XSTRM-36): The objects in the stream layer-datatype '94:22' are ignored. This is because the stream layer-datatype pair is not specified in the layer map file. Specify the stream layer-datatype pair in the layer map file and try again.
WARNING (XSTRM-36): The objects in the stream layer-datatype '95:0' are ignored. This is because the stream layer-datatype pair is not specified in the layer map file. Specify the stream layer-datatype pair in the layer map file and try again.
```

msaligane, 2020-07-22:
@mithro After checking this https://docs.google.com/spreadsheets/d/176OcP2h_TT_pDP2-zBLijlgFUP38LVgMQPlDtpLcrsk/edit?ts=5eff8387&pli=1#gid=881680160
I guess this issue is related to  #91

RTimothyEdwards, 2020-07-22:
@mithro, @msaligane :  To make this work right, all of the technology LEF files need to be updated to define MASTERSLICE entries for "nwell" and "pwell" layers.  I can do this update in the .tlef files in skywater-src-nda branch "filtered" directory "tlef2".  If that sounds okay to you, I will go ahead and do that and commit.

The other changes that need to be made (which may or may not be reflected in the source repos yet) are:

(1) All (digital standard cell library) LEF files (preferably in the original sources from which everything else is derived) must move "VNB" from layer "li1" or "met1" and place it on layer "pwell".  "VPB" must be moved from layer "li1" or "met1" and placed on layer "nwell".  This is the root cause of all the problems we've been having, because it is just dead wrong.  Those pins do NOT exist on the metal layers, and putting them there just causes the tools to assume they are shorted to power and ground, which they are decidedly not.

(2) The technology lef files must have MASTERSLICE entries for "nwell" and "pwell".

mithro, 2020-07-22:
@RTimothyEdwards Please update the tlef files update in the .tlef files in skywater-src-nda branch "filtered" directory "tlef2". 

I will look at updating the LEF files to have;
 - `VNB` pin are placed on the layer `nwell`, and
 - `VPB` pin are placed on the layer `pwell`.

RTimothyEdwards, 2020-07-22:
@msaligane :  Okay, I had not looked at some of the most recent messages, so I didn't see the comments above about the layer map.  I looked through the file sky130_fd_sc_hs__nor2_1.gds, and those three layers mentioned (92:0, 94:22, and 95:0) do not exist (those layers are also unknown by the Calibre DRC and LVS decks).

RTimothyEdwards, 2020-07-22:
@mithro :  I am working on the updated entries now;  pulling recent changes, then I will commit.

mithro, 2020-07-22:
@RTimothyEdwards - I moved the discussion around the MASTERSLICE to #121 and I'll leave this bug open around the GDS layers and Cadence.

msaligane, 2020-07-22:
@RTimothyEdwards strange.. Those layers seems to be required to pass LVS but now by just adding pin+label to vnb/vpb, the LVS passes (without having them - I made a new from scratch). So I think this could be ignored for now, till I run LVS on a larger design.