google/skywater-pdk #121: Technology LEF files need to be updated to define MASTERSLICE entries for "nwell" and "pwell" layers

mithro, 2020-07-22:
_Originally posted by @RTimothyEdwards in https://github.com/google/skywater-pdk/issues/114#issuecomment-662566186_
> @mithro, @msaligane :  To make this work right, all of the technology LEF files need to be updated to define MASTERSLICE entries for "nwell" and "pwell" layers.  I can do this update in the .tlef files in skywater-src-nda branch "filtered" directory "tlef2".  If that sounds okay to you, I will go ahead and do that and commit.
> 
> The other changes that need to be made (which may or may not be reflected in the source repos yet) are:
> 
> (1) All (digital standard cell library) LEF files (preferably in the original sources from which > everything else is derived) must move "VNB" from layer "li1" or "met1" and place it on layer "pwell".  "VPB" must be moved from layer "li1" or "met1" and placed on layer "nwell".  > This is the root cause of all the problems we've been having, because it is just dead wrong.  Those pins do NOT exist on the metal layers, and putting them there just causes  the tools to assume they are shorted to power and ground, which they are decidedly not.
>
> (2) The technology lef files must have MASTERSLICE entries for "nwell" and "pwell".
>

mithro, 2020-07-22:
Originally posted by me in https://github.com/google/skywater-pdk/issues/114#issuecomment-662708814
> @RTimothyEdwards Please update the tlef files update in the .tlef files in skywater-src-nda branch "filtered" directory "tlef2".
> 
> I will look at updating the LEF files to have;
> 
> * `VNB` pin are placed on the layer `nwell`, and
> * `VPB` pin are placed on the layer `pwell`.

mithro, 2020-07-22:
@RTimothyEdwards What should the following LEF be? It has `vpb` and `destvpb` pins. The pins also have multiple ports on the `li1` layer?
```
MACRO scs8lp_lsbuf_lp
  CLASS CORE ;
  SOURCE USER ;
  ORIGIN 0 0 ;
  SIZE 4.8 BY 6.66 ;
  SYMMETRY X Y R90 ;
  SITE unit ;

  PIN A
    DIRECTION INPUT ;
    USE SIGNAL ;
    PORT
      LAYER li1 ;
        RECT 0.5850 1.5300 1.2450 2.1350 ;
        RECT 1.0750 2.1350 1.2450 2.7750 ;
        RECT 1.0750 2.7750 1.9100 3.0750 ;
    END
    ANTENNAGATEAREA 0.678 LAYER met1 ;
    ANTENNAGATEAREA 0.678 LAYER met2 ;
  END A

  PIN X
    DIRECTION OUTPUT ;
    USE SIGNAL ;
    PORT
      LAYER li1 ;
        RECT 3.8950 3.5850 4.2250 6.4050 ;
    END
    ANTENNADIFFAREA 0.4876 LAYER met1 ;
    ANTENNADIFFAREA 0.4876 LAYER met2 ;
    ANTENNADIFFAREA 0.4876 LAYER met3 ;
  END X

  PIN destvpb
    DIRECTION INOUT ;
    USE POWER ;
    PORT
      LAYER li1 ;
        RECT 4.4100 5.2200 4.7100 6.3950 ;
    END
    PORT
      LAYER li1 ;
        RECT 0.0900 5.2200 0.3900 6.3950 ;
    END
  END destvpb

  PIN vpb
    DIRECTION INOUT ;
    USE POWER ;
    PORT
      LAYER li1 ;
        RECT 4.4100 0.2650 4.7100 1.4400 ;
    END
    PORT
      LAYER li1 ;
        RECT 0.0900 0.2650 0.3900 1.4400 ;
    END
  END vpb

  PIN vgnd
    DIRECTION INOUT ;
    USE GROUND ;
    PORT
      LAYER met1 ;
        RECT 0.0000 3.0850 4.8000 3.5750 ;
    END
  END vgnd

  PIN destpwr
    DIRECTION INOUT ;
    USE POWER ;
    PORT
      LAYER met1 ;
        RECT 0.0000 6.4150 4.8000 6.9050 ;
    END
  END destpwr

  PIN vpwr
    DIRECTION INOUT ;
    USE POWER ;
    PORT
      LAYER met1 ;
        RECT 0.0000 -0.2450 4.8000 0.2450 ;
    END
  END vpwr
  OBS
    LAYER li1 ;
      RECT 2.5050 5.5250 2.8350 6.3950 ;
      RECT 1.5100 5.3550 2.8350 5.5250 ;
      RECT 2.6650 5.3300 2.8350 5.3550 ;
      RECT 1.5100 5.2650 1.6800 5.3550 ;
      RECT 2.6650 5.1600 3.1600 5.3300 ;
      RECT 1.3500 4.9900 1.6800 5.2650 ;
      RECT 2.9900 4.5700 3.1600 5.1600 ;
      RECT 2.7650 4.4400 3.1600 4.5700 ;
      RECT 2.5500 4.4000 3.1600 4.4400 ;
      RECT 2.5500 4.1400 2.9350 4.4000 ;
      RECT 2.5050 3.5850 2.9350 4.1400 ;
      RECT 0.0000 6.5750 4.8000 6.7450 ;
      RECT 1.7150 5.6950 2.0450 6.5750 ;
      RECT 3.1050 5.5000 3.4350 6.5750 ;
      RECT 0.0000 3.2450 1.9950 3.4150 ;
      RECT 1.7150 3.4150 1.9950 4.4800 ;
      RECT 0.0900 3.4150 0.3900 4.7700 ;
      RECT 0.0900 1.8900 0.3900 3.2450 ;
      RECT 0.5750 2.3050 0.9050 3.2450 ;
      RECT 0.9250 5.4350 1.2750 6.4050 ;
      RECT 0.9250 4.8200 1.1800 5.4350 ;
      RECT 0.9250 4.6500 2.0400 4.8200 ;
      RECT 1.8700 4.8200 2.0400 5.0100 ;
      RECT 0.9250 3.5850 1.2050 4.6500 ;
      RECT 1.8700 5.0100 2.2800 5.1850 ;
      RECT 2.5050 3.2450 4.8000 3.4150 ;
      RECT 4.4100 3.4150 4.7100 4.7700 ;
      RECT 4.4100 1.8900 4.7100 3.2450 ;
      RECT 3.1050 3.4150 3.4350 4.2300 ;
      RECT 2.4500 4.8400 2.8200 4.9900 ;
      RECT 2.2100 4.7400 2.8200 4.8400 ;
      RECT 2.2100 4.6100 2.5950 4.7400 ;
      RECT 2.2100 4.4800 2.3800 4.6100 ;
      RECT 2.1650 4.3100 2.3800 4.4800 ;
      RECT 2.1650 2.6050 2.3350 4.3100 ;
      RECT 1.4150 2.4350 2.3350 2.6050 ;
      RECT 1.4150 1.3600 1.7150 2.4350 ;
      RECT 1.3650 0.2650 1.7150 1.3600 ;
      RECT 0.0000 -0.0850 4.8000 0.0850 ;
      RECT 0.5750 0.0850 0.9050 1.1750 ;
    LAYER mcon ;
      RECT 2.0750 6.5750 2.2450 6.7450 ;
      RECT 1.5950 -0.0850 1.7650 0.0850 ;
      RECT 1.5950 3.2450 1.7650 3.4150 ;
      RECT 1.5950 6.5750 1.7650 6.7450 ;
      RECT 1.1150 -0.0850 1.2850 0.0850 ;
      RECT 1.1150 3.2450 1.2850 3.4150 ;
      RECT 1.1150 6.5750 1.2850 6.7450 ;
      RECT 0.6350 6.5750 0.8050 6.7450 ;
      RECT 3.0350 6.5750 3.2050 6.7450 ;
      RECT 0.1550 -0.0850 0.3250 0.0850 ;
      RECT 3.0350 3.2450 3.2050 3.4150 ;
      RECT 3.0350 -0.0850 3.2050 0.0850 ;
      RECT 3.5150 6.5750 3.6850 6.7450 ;
      RECT 3.5150 3.2450 3.6850 3.4150 ;
      RECT 3.5150 -0.0850 3.6850 0.0850 ;
      RECT 3.9950 6.5750 4.1650 6.7450 ;
      RECT 3.9950 3.2450 4.1650 3.4150 ;
      RECT 3.9950 -0.0850 4.1650 0.0850 ;
      RECT 4.4750 6.5750 4.6450 6.7450 ;
      RECT 4.4750 3.2450 4.6450 3.4150 ;
      RECT 4.4750 -0.0850 4.6450 0.0850 ;
      RECT 0.6350 3.2450 0.8050 3.4150 ;
      RECT 0.6350 -0.0850 0.8050 0.0850 ;
      RECT 0.1550 6.5750 0.3250 6.7450 ;
      RECT 0.1550 3.2450 0.3250 3.4150 ;
      RECT 2.5550 -0.0850 2.7250 0.0850 ;
      RECT 2.5550 3.2450 2.7250 3.4150 ;
      RECT 2.5550 6.5750 2.7250 6.7450 ;
      RECT 2.0750 -0.0850 2.2450 0.0850 ;
  END
END scs8lp_lsbuf_lp
```

RTimothyEdwards, 2020-07-22:
@mithro : I just pushed the update to the technology LEF files.  I will check the cell above and let you know what I determine about it.

RTimothyEdwards, 2020-07-22:
"destvpb" and "vpb" are treated the same.  This is a double-height cell, so "destvpb" is the "vpb" connection on top, and "vpb" is the one on the bottom.

Unfortunately, this is a case where "vpb" is in fact connected to li1.  So it seems impossible to automatically determine whether or not it is connected to a real layer, other than to note when the port overlaps another port on the same layer.

The two ports is a correct use of "PORT" in this case.  There are two ports, on either side of the cell, and they are connected only through a high-resistance path (the nwell).

kevink78, 2020-07-23:
Sorry, I was too busy to respond to these while they were coming by. vpb and vnb are secondary power pins, and they are on well layers. It can be problematic for LEF. At some point you mentioned no TIEHI/TIELO cells. There should be a cell called "conb" (constant generator) which has a tie hi and tie lo pin. But I agree, a cell with a more obvious name makes sense.

RTimothyEdwards, 2020-07-23:
@mithro :  I had a typo in yesterday's commit to skywater-src-nda "filtered" branch and just pushed the correction.

mithro, 2020-09-28:
Has this been done?

RTimothyEdwards, 2020-09-28:
@mithro :  Apparently not?  I am no longer sure what my comment of July 22 was referring to.

RTimothyEdwards, 2020-09-28:
@mithro :  My comments in the git log of July 22 say that I added MASTERSLICE to all of the technology LEF files.  However, there is no evidence of this.  Did those changes get overwritten?

RTimothyEdwards, 2020-09-28:
@mithro :  Ah, right, there is a directory skywater-src-nda/tlef2/ with the corrected tech files.  But I don't see this reflected in the sky130_fd_sc_* files.

PR #164 (2020-10-05): Fixing the technology LEF files.
The technology LEF file was not updated when https://github.com/google/skywater-pdk/pull/155 was merged.

Technology LEF file changes include;
 * Adding `PROPERTYDEFINITIONS`
 * Setting `nwell` and `pwell` to `MASTERSLICE` type.
 * Fixing the capacitance values for the various layers.

This fixes #121.

The submodule updates are;
 - Updating [`sky130_fd_sc_hd` latest](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/v0.0.2-32-g300eb6fb..v0.0.2-34-ga01b5960) to v0.0.2-34-ga01b5960
 - Updating [`sky130_fd_sc_hd` v0.0.1](https://foss-eda [... cut]
closed 2020-10-05 by mithro