google/skywater-pdk #122: Rework the tie cells in the PDK

mithro, 2020-07-22:
It was pointed out in https://github.com/google/skywater-pdk/issues/113 that the LEF files don't have `TIELOW` and `TIEHIGH` cells in the LEF at the moment.

The `conb` cell was potentially indicated as being a tie cell in https://github.com/google/skywater-pdk/issues/113#issuecomment-662001448 but it seems to connect to both `TIELOW` and `TIEHIGH` and LEF does not support the specification of both together.

@RTimothyEdwards suggested in https://github.com/google/skywater-pdk/issues/113#issuecomment-662561845 that we could create two copies of the `conb` cell with one being marked as a `TIELOW` and one marked as `TIEHIGH`.

@msaligane mentioned in https://github.com/google/skywater-pdk/issues/113#issuecomment-662723020
> @RTimothyEdwards It would really make sense to invest in making two new cells for the tie hi/lo. One other thing that really bothers me is the resistor used there. I would expect a pull-up / pull-down circuit. Something that looks like this: https://electronics.stackexchange.com/questions/195887/basic-tie-high-and-tie-low-circuits-in-digital-vlsi-design
> 
> As @rovinski mentioned, not a very conventional approach.

mithro, 2020-07-22:
@msaligane / @rovinski -- Any chance you are would be interested in looking into solving this problem?

msaligane, 2020-07-22:
@mithro meaning creating these two cells? It is doable but I ll need about 2 weeks or so to get to it..

mithro, 2020-07-22:
I'll leave this open for anyone until you have time to take it on then.

20Mhz, 2020-07-23:
Just a note, different versions will be required match each library type

RTimothyEdwards, 2020-07-23:
@msaligane :  I would call the use of a resistor very standard for this kind of cell.  You want the cell's output to have a moderately low resistance to the power or ground rail, but not too low, because that's equivalent to tying the output directly to the rail, which then effectively ties all gates downstream of the cell to the rail as well, which makes those gates very sensitive to ESD events.  The output of the tie cell is NOT expected to draw any DC current.

RTimothyEdwards, 2020-07-23:
@mithro :  I can create the two new cells per library.  It is relatively trivial to do.  My question is:  where do you want me to put the new cells?  Should they be formatted like the original skywater-src-nda, or formatted like the final skywater-pdk?

mithro, 2020-07-28:
Whichever is easiest for you.