google/skywater-pdk #126: Make sure the PDK can be used without conda if you supply a valid environment

mithro, 2020-07-23:
For ease of use a prebuilt conda environment is provided. 

For advanced users who do not wish to use this pre-built environment, it should still be possible to use all the Makefile targets (the `$(IN_CONDA_ENV)` should just evaluate to nothing.

The environment should also be checking the dependencies that are normally provided by conda are installed and available.

mithro, 2020-07-23:
@smunaut -- As the primary person I know who dislikes conda, can you see if this currently works and if not what is required to make it work?

mithro, 2020-07-23:
This is related to #46 

On a somewhat related note, it would also be good if [`make-env`](https://github.com/SymbiFlow/make-env) support more "environment types" like;
 - virtualenv + system binaries - https://github.com/SymbiFlow/make-env/issues/16
 - docker - https://github.com/SymbiFlow/make-env/issues/15

smunaut, 2020-07-23:
ATM I define :

```
CONDA_ENV_PYTHON=
IN_CONDA_ENV=
```

and don't include `conda.mk` and this works fine for me.

proppy, 2022-03-24:
+1 ideally the `Makefile` should not make any assumption on how things are installed to make it possible for advanced users, packagers and downstream dependencies to manage dependencies by themselves; if we need some convenient way to provision a build environment that should be done/documented from outside of the core build system.