google/skywater-pdk #128: What is the recommended amount of disk space?

galv, 2020-07-24:
When I did `git clone --recursive` on this repo, I actually ran out of disk space on a disk with ~25GB free. How large a disk do you recommend for downloading this PDK?

RTimothyEdwards, 2020-07-24:
@galv :  Looking at my own disk space used by skywater-pdks, the vast majority is used by the libraries (and the majority of that is used by the liberty files).  The rest of the PDK is about 1.5GB.  I pulled only the "latest" versions of each of the digital repositories, not all versions, and the total used after running "make timing" is 13.8GB for the libraries (so 15.3GB total or so).  My advice is that you not try to download all versions of all libraries.  Unless you have some need for every digital standard cell library, I also suggest that you download only the libraries you need.  Start with sky130_fd_sc_hd, since that is the one that has been tested the most with the openlane digital flow.

mithro, 2020-07-24:
Looks like the PDK repo with all the libraries initialized is roughly ~50 gigabytes.

20Mhz, 2020-07-28:
Btw, only about 2G then downloading and building only latest sky130_fd_sc_hd, enough for full open lane run

mithro, 2020-07-28:
Someone want to add a section to the README and the documentation?