google/skywater-pdk #134: Remove UNITS from cells LEF files

msaligane, 2020-07-25:
## Expected Behavior
This value should be defined only in the tech LEF as 2000. 

## Actual Behavior
This is defined inside each of the cells, however, it should only be defined in the tech lef.
```
UNITS
  DATABASE MICRONS 200 ;
```

Innovus reports this warning:
`The 'UNITS' attribute is only allowed to be set in the first LEF file (technology LEF). The attribute was found in this LEF file, and will be ignored.`

When importing the design in virtuoso, the track height is larger than the cells height (the cell size is correct). See below:

[image]

RTimothyEdwards, 2020-07-25:
@msaligane :  Shouldn't that be "DATABASE MICRONS 200"?  LEF units are microns, so 1 database unit = 1 micron / 200 = 5nm, which is the minimum grid.

RTimothyEdwards, 2020-07-25:
@mithro :  I pushed another patch to magic to remove the "UNITS DATABASE MICRONS" line from all LEF files that do not include the technology LEF portion.

mithro, 2020-07-25:
@msaligane -- For future pull requests, can you mention if this applies to `.magic.lef`, `.lef` or both?

msaligane, 2020-07-25:
@RTimothyEdwards 

> Shouldn't that be "DATABASE MICRONS 200"? LEF units are microns, so 1 database unit = 1 micron / 200 = 5nm, which is the minimum grid.

How about the tech LEF? The DBU is 1000. It doesn't makes sense to me to have different units for each LEF (cell vs tech lef)
So I think this should be removed. After, fixing this manually, my gds seems to have the correct geometry. 

> @msaligane -- For future pull requests, can you mention if this applies to .magic.lef, .lef or both?

Sure, I only see this in magic.lef

mithro, 2020-09-28:
This was done in #155

closed 2020-09-28 by mithro