google/skywater-pdk #138: Set up slack channel invite automation

mithro, 2020-07-29:
Set up https://github.com/outsideris/slack-invite-automation to allow people to automatically invite themselves to [the skywater-pdk slack](https://skywater-pdk.slack.com).

Should also make it pretty and match the material design theme @ [the skywater-pdk documentation](https://skywater-pdk.readthedocs.io/en/latest/).

Should include links back to the GitHub / Documentation / etc. See example of the links from the SymbiFlow page;
[image]

mithro, 2020-07-29:
@mkkassem is working on this.

mithro, 2020-07-29:
Should also make sure that the slack channel is linked to from;
 - [ ] the README.rst file
 - [ ] the SkyWater PDK documentation
 - [ ] On the slack channel
 - [ ] Sent to the announcement list

mithro, 2020-08-04:
https://invite.skywater.pdk and https://join.skywater.pdk URLs should now take you to the slack auto-inviter thingy.

https://slack.skywater.pdk should take you to the slack workspace.

matthewpang, 2020-08-04:
@mithro DNS not resolving to skywater.pdk unfortunately

mithro, 2020-08-04:
It should be skywater.tools not skywater.pdk (updated my comment) -- guess I need get more sleep.

 * https://invite.skywater.tools and https://join.skywater.tools 
 * https://slack.skywater.tools

mithro, 2020-08-06:
@mkkassem - Could you look at getting the documentation updated with the links to the skywater slack?

svenn71, 2021-11-14:
Trying to invite myself today, but got the message that this site can't be reached.

mithro, 2021-11-14:
FYI - @mkkassem

mkkassem, 2021-11-15:
@svenn71  please try now. There was a problem with DNS record. Corrected now.

svenn71, 2021-11-15:
@mkkassem I was now able to join the skywater-pdk, thank you.