google/skywater-pdk #141: sky130_fd_sc_hs/latest random LEF issues in fill_diode and diode cells

msaligane, 2020-08-11:
## Expected Behavior
The following is the correction for each issue: 

1. should be `CLASS CORE ANTENNACELL ;`
2. add  `SYMMETRY X Y ; ` which is missing in `cells/diode/sky130_fd_sc_hs__diode_2.magic.lef`
3. should be `CLASS SPACER ;`
4. add  `SYMMETRY X Y ; ` which is missing in `cells/fill_diode/*.magic.lef`
5. self explanatory - Both the normal LEFs and the magic LEFs for diode and fill_diode in `sky130_fd_sc_hs/v0.0.1`  are correct. So maybe reuse that?

## Actual Behavior
These are a couple issues seen inside the magic LEF of the fill_diode and diode cells:

1.  `CLASS BLOCK ;`  is incorrect in `cells/diode/sky130_fd_sc_hs__diode_2.magic.lef`
2.  `SYMMETRY X Y ; ` is missing in `cells/diode/sky130_fd_sc_hs__diode_2.magic.lef`
3.  `CLASS BLOCK ;` is missing in `cells/fill_diode/*.magic.lef`
4.  `SYMMETRY X Y ;` is missing in `cells/fill_diode/*.magic.lef`
5.  Normal LEFs for diode and fill_diode are missing in `sky130_fd_sc_hs/latest` and `sky130_fd_sc_hs/v0.0.2`

mithro, 2020-08-11:
Are you saying that `CLASS BLOCK` should be `CLASS CORE ANTENNACELL ;` in `cells/diode/sky130_fd_sc_hs__diode_2.magic.lef`?

msaligane, 2020-08-11:
> Are you saying that CLASS BLOCK should be CLASS CORE ANTENNACELL ; in cells/diode/sky130_fd_sc_hs__diode_2.magic.lef?
Correct

agorararmard, 2020-09-23:
@mithro @msaligane This is the same exact case with the sky130_fd_sc_ms. The same files have the same issues with the same fixes.

So, the description of the diode cell should be for the hs cell: cells/diode/sky130_fd_sc_hs__diode_2.magic.lef
```
MACRO sky130_fd_sc_hs__diode_2
  CLASS CORE ANTENNACELL ;                  -->> This is new
  FOREIGN sky130_fd_sc_hs__diode_2 ;
  ORIGIN  0.000000  0.000000 ;
  SIZE  0.960000 BY  3.330000 ;
  SYMMETRY X Y ;                   -->> This is new
  SITE unit ;                   -->> This is new

  PIN DIODE
    ANTENNADIFFAREA  0.641700 ;
    DIRECTION INPUT ;                  -->> This is new and nice to have
    USE SIGNAL ;                  -->> This is new and nice to have
    PORT
      LAYER li1 ;
        RECT 0.095000 0.265000 0.865000 3.065000 ;
    END
  END DIODE
...... and so on.
```
The same lines should be added for cells/diode/sky130_fd_sc_ms__diode_2.magic.lef

RTimothyEdwards, 2020-09-24:
@agorararmard :  The problem is that these are cells that didn't have a LEF view in SkyWater, and magic doesn't automatically know how to categorize them, so the SITE, SYMMETRY, CLASS, and pin DIRECTION and USE all have to be manually added.  I'd assume that this is something that is best to be just applied as a patch to the repository.

mithro, 2020-10-30:
Do we have corrected `.lef` files for these diodes around somewhere?

msaligane, 2020-10-30:
@agorararmard Looks good to me!

rovinski, 2020-10-31:
@agorararmard I think the files are missing the changes made in #196 and some other things:

```
  PIN VNB                                          # Missing USE GROUND
    DIRECTION INPUT ;                              # Should be INOUT
    PORT
      LAYER pwell ;
        RECT 0.000000 0.000000 0.960000 0.245000 ;
    END
  END VNB
  PIN VPB                                          # Missing USE POWER
    DIRECTION INPUT ;                              # Should be INOUT
    PORT
      LAYER nwell ;
        RECT -0.190000 1.660000 1.150000 3.520000 ;
    END
  END VPB
  PIN VGND                                         # Missing SHAPE ABUTMENT
    DIRECTION INPUT ;                              # Should be INOUT
    USE GROUND ;
    PORT
      LAYER met1 ;
        RECT 0.000000 -0.245000 0.960000 0.245000 ;
    END
  END VGND
  PIN VPWR                                         # Missing SHAPE ABUTMENT
    DIRECTION INPUT ;                              # Should be INOUT
    USE POWER ;
    PORT
      LAYER met1 ;
        RECT 0.000000 3.085000 0.960000 3.575000 ;
    END
  END VPWR

```

agorararmard, 2020-11-02:
@rovinski: Great catch! I actually overlooked this because none of the tools showed any errors. Interestingly enough, this **is** using #196 + modifications on the diode pin and macro info. So in #196 these 2 cells have inaccurate power pins descriptions.

I'm gonna check to see if this issue is exclusive to these 2 cells or not.

agorararmard, 2020-11-02:
@mithro: so, there you go again with the corrected power pins: [diodes.tar.gz](https://github.com/google/skywater-pdk/files/5475607/diodes.tar.gz)

@rovinski: this covers it, right?

agorararmard, 2020-11-02:
@mithro: The following cells slipped throw #196 unfixed:
ls:
```
sky130_fd_sc_ls__decaphetap_2
```
ms:
```
sky130_fd_sc_ms__diode_2
sky130_fd_sc_ms__fill_diode_2
sky130_fd_sc_ms__fill_diode_4
sky130_fd_sc_ms__fill_diode_8
```
hs:
```
sky130_fd_sc_hs__diode_2
sky130_fd_sc_hs__fill_diode_2
sky130_fd_sc_hs__fill_diode_4
sky130_fd_sc_hs__fill_diode_8
```

Should I open a separate issue for them, or should we consider this a follow up to this issue since they are mostly diodes?

mithro, 2020-11-02:
@agorararmard - Can you commit them to a github repo (a gist.github.com is suitable if you just need a temporary)?

mithro, 2020-11-02:
(And then link them here.)

agorararmard, 2020-11-02:
@mithro: https://gist.github.com/agorararmard/4e2a83d31bccc451d48e72928212f5de
I added the fill_diodes as well according to the notes @msaligane gave in the body of the issue `CLASS SPACER` and `SYMMETRY X Y`.

rovinski, 2020-11-02:
@agorararmard looks good to me.

msaligane, 2020-11-02:
@agorararmard Yes, looks good. I think the previous comment were reported in a different issue. Thanks!

mithro, 2020-11-03:
Do these same issues appear for the other standard cell libraries? Do these have the same issues?
 - [ ] `sky130_fd_sc_ls`
 - [ ] `sky130_fd_sc_lp`
 - [ ] `sky130_fd_sc_hd`
 - [ ] `sky130_fd_sc_hdll`

mithro, 2020-11-03:
It seems like the `sky130_fd_sc_ls` does?

agorararmard, 2020-11-03:
> Do these same issues appear for the other standard cell libraries? Do these have the same issues?
> 
>     * [ ]  `sky130_fd_sc_ls`
> 
>     * [ ]  `sky130_fd_sc_lp`
> 
>     * [ ]  `sky130_fd_sc_hd`
> 
>     * [ ]  `sky130_fd_sc_hdll`

@mithro: Nope, all look good to me.

The `sky130_fd_sc_ls__decaphetap_2` cell is the only case I came across that has an incorrect description for power pins (wasn't corrected in #196). But, the diodes look good and don't flag errors/warnings in the flow as well.

I'll let you know if anything came up.

mithro, 2020-11-04:
@agorararmard - The pull request at https://github.com/google/skywater-pdk/issues/141 should fix the issue with the diodes.

agorararmard, 2020-11-16:
@mithro: it seems that you used an older revision of the gist (there is a few hours difference). The only difference is replacing `CLASS SPACER` with `CLASS CORE SPACER`, because `CLASS SPACER` is not valid LEF syntax. This is for the `fill_diode` cells.

Here is the gist again for your convenience: https://gist.github.com/agorararmard/4e2a83d31bccc451d48e72928212f5de

agorararmard, 2020-11-16:
@mithro: Should I open a separate issue or should we re-open this one?

mithro, 2020-11-17:
@agorararmard - I believe https://github.com/google/skywater-pdk/pull/243 should fix the issue.

closed 2020-11-17 by mithro