google/skywater-pdk #146: Cleanup references to older flows introduced in pull request #127.

mithro, 2020-09-14:
When https://github.com/google/skywater-pdk/pull/127 was merged there was a bunch of references to the older proprietary flows (see [previous names](https://skywater-pdk.readthedocs.io/en/latest/previous.html)). 

These need to be cleaned up and removed.

diadatp, 2020-11-07:
I did a quick look through and found the following references that aren't explained in previous.rst:

S8P12-10R
S8PF-10R
S8Q-5R
S8T
S8TM-5R
S8TMA-5R
S8TMC-5R
S8TNV-5R
SKY130P
SKY130T
SP8P-5R
SP8Q
SP8TEE-5R

- It might be worthwhile to explain a few of these in previous.rst
- Some of these occurrences deal with options. Is it safe to assume that all options are available and drop the references for the more restricted processes? For example [here](https://github.com/google/skywater-pdk/blob/38a749752ca85738b9ce56b948cafc139ed8423b/docs/rules/layers/table-c3-device-lvs-other.csv#L51) and [here](https://github.com/google/skywater-pdk/blob/38a749752ca85738b9ce56b948cafc139ed8423b/docs/rules/layers/table-f2a-lvs.tsv#L54).

mithro, 2020-11-07:
@diadatp - I don't think **all** options are available, just most. I think @RTimothyEdwards had figured out the various overlaps previously?

RTimothyEdwards, 2020-11-07:
There are no options in sky130.  All of these references are outdated and not useful (and confusing).  Those that refer to top metal as m3 vs. m5 are the easiest to eliminate---there are five metal layers and m5 is top and that's it.

For a start, "T", "Q", "P" stands for 3, 4, and 5 metal layers, so only "P" matches sky130.  5R and 10R refer to the highest voltage support (not counting extended drain devices), so 5R is out.  I don't remember what "F" refers to in the above nomenclature.  In the usual nomenclature, this process is closest to s8phirs-10r, but that doesn't exactly match any of the above.