google/skywater-pdk #152: Verilog files cannot use "cell" as an instance name

RTimothyEdwards, 2020-09-25:
All of the verilog standard cells are instancing the root module for the standard cell as <cellname> cell ( ... ).  Apparently "cell" is a reserved keyword in verilog?  At any rate, iverilog throws an inscrutable error and refuses to simulate.  Any reasonable substitution works, e.g., "sc_cell" or somesuch.

fangism, 2020-09-30:
From my notes in verilog.y:

```yacc
/* The 1364-2001 configuration tokens. */
%token TK_cell "cell"
```

mkkassem, 2020-09-30:
### OBSERVED

I noticed that the "cells" string is used in cells with multiple drive strengths. 
The base cell has a test bench but the _1, _2, _4 versions don't have test benches that explicitly instantiates them.
The available test bench instantiates the base cell - only.

sky130_fd_sc_hd__a22oi.v
sky130_fd_sc_hd__a22oi.tb.v
.....
`sky130_fd_sc_hd__a22oi dut (.A1(A1), .A2(A2), .B1(B1), .B2(B2), .VPWR(VPWR), .VGND(VGND), .VPB(VPB), .VNB(VNB), .Y(Y));
` 
### EXPECTED

if indeed there is a simulation test run with the ci then we should have a test bench `sky130_fd_sc_hd__a22oi_1.tb.v` that instantiates the cell including the drive strength suffix.

`sky130_fd_sc_hd__a22oi_1 dut (.A1(A1), .A2(A2), .B1(B1), .B2(B2), .VPWR(VPWR), .VGND(VGND), .VPB(VPB), .VNB(VNB), .Y(Y));
`

PR #166 (2020-10-06): verilog: Fixing usage of cell reserved word.
`cell` is a Verilog reserved word.

Fixes #152.

 - Updating [`sky130_fd_sc_hd` latest](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/v0.0.2-36-gc70c051b..v0.0.2-38-g9b8d061b) to v0.0.2-38-g9b8d061b
 - Updating [`sky130_fd_sc_hd` v0.0.1](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/v0.0.1-18-gfa8d392b6..v0.0.1-19-g23c9281bd) to v0.0.1-19-g23c9281bd
 - Updating [`sky130_fd_sc_hd` v0.0.2](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/v0.0.2-36-gc70c051b..v0.0.2- [... cut]
closed 2020-10-06 by mithro