google/skywater-pdk #153: Verilog primitives do not start with the output signal

RTimothyEdwards, 2020-09-25:
@mithro : According to iverilog output, the first signal in a primitive's pin list must be the output.

Therefore entries like sky130_fd_sc_hd/latest/models/udp_mux_2to1_n/sky130_fd_sc_hd__udp_mux_2to1_n.v are incorrect; e.g.,

    primitive sky130_fd_sc_hd__udp_mux_2to1_N ( A0, A1, S, Y );

should instead be

   primitive sky130_fd_sc_hd__udp_mux_2to1_N ( Y, A0, A1, S );

In fact, the calls to the primitives are in a different order than the primitive, which is completely wrong;  e.g., sky130_fd_sc_hd__mux2i.functional.v has:

   sky130_fd_sc_hd__udp_mux_2to1_N mux_2to1_n0 ( mux_2to1_n0_out_Y, A0, A1, S );

So the primitives are not even internally consistent with their instances.

mithro, 2020-10-06:
As far as I can see, this only affects the mux primitives;
```
 libraries/sky130_fd_sc_hd/v0.0.1/models/udp_mux_2to1_n/sky130_fd_sc_hd__udp_mux_2to1_n.blackbox.v     | 6 +++---
 libraries/sky130_fd_sc_hd/v0.0.1/models/udp_mux_2to1_n/sky130_fd_sc_hd__udp_mux_2to1_n.v              | 6 +++---
 libraries/sky130_fd_sc_hd/v0.0.1/models/udp_mux_4to2/sky130_fd_sc_hd__udp_mux_4to2.blackbox.v         | 6 +++---
 libraries/sky130_fd_sc_hd/v0.0.1/models/udp_mux_4to2/sky130_fd_sc_hd__udp_mux_4to2.v                  | 6 +++---
 libraries/sky130_fd_sc_hd/v0.0.2/models/udp_mux_2to1_n/sky130_fd_sc_hd__udp_mux_2to1_n.blackbox.v     | 6 +++---
 libraries/sky130_fd_sc_hd/v0.0.2/models/udp_mux_2to1_n/sky130_fd_sc_hd__udp_mux_2to1_n.v              | 6 +++---
 libraries/sky130_fd_sc_hd/v0.0.2/models/udp_mux_4to2/sky130_fd_sc_hd__udp_mux_4to2.blackbox.v         | 6 +++---
 libraries/sky130_fd_sc_hd/v0.0.2/models/udp_mux_4to2/sky130_fd_sc_hd__udp_mux_4to2.v                  | 6 +++---
 libraries/sky130_fd_sc_hdll/v0.1.0/models/udp_mux_2to1_n/sky130_fd_sc_hdll__udp_mux_2to1_n.blackbox.v | 6 +++---
 libraries/sky130_fd_sc_hdll/v0.1.0/models/udp_mux_2to1_n/sky130_fd_sc_hdll__udp_mux_2to1_n.v          | 6 +++---
 libraries/sky130_fd_sc_hdll/v0.1.0/models/udp_mux_4to2/sky130_fd_sc_hdll__udp_mux_4to2.blackbox.v     | 6 +++---
 libraries/sky130_fd_sc_hdll/v0.1.0/models/udp_mux_4to2/sky130_fd_sc_hdll__udp_mux_4to2.v              | 6 +++---
 libraries/sky130_fd_sc_hdll/v0.1.1/models/udp_mux_2to1_n/sky130_fd_sc_hdll__udp_mux_2to1_n.blackbox.v | 6 +++---
 libraries/sky130_fd_sc_hdll/v0.1.1/models/udp_mux_2to1_n/sky130_fd_sc_hdll__udp_mux_2to1_n.v          | 6 +++---
 libraries/sky130_fd_sc_hdll/v0.1.1/models/udp_mux_4to2/sky130_fd_sc_hdll__udp_mux_4to2.blackbox.v     | 6 +++---
 libraries/sky130_fd_sc_hdll/v0.1.1/models/udp_mux_4to2/sky130_fd_sc_hdll__udp_mux_4to2.v              | 6 +++---
 libraries/sky130_fd_sc_hs/v0.0.0/models/udp_mux_2to1_n/sky130_fd_sc_hs__udp_mux_2to1_n.blackbox.v     | 6 +++---
 libraries/sky130_fd_sc_hs/v0.0.0/models/udp_mux_2to1_n/sky130_fd_sc_hs__udp_mux_2to1_n.v              | 6 +++---
 libraries/sky130_fd_sc_hs/v0.0.0/models/udp_mux_4to2/sky130_fd_sc_hs__udp_mux_4to2.blackbox.v         | 6 +++---
 libraries/sky130_fd_sc_hs/v0.0.0/models/udp_mux_4to2/sky130_fd_sc_hs__udp_mux_4to2.v                  | 6 +++---
 libraries/sky130_fd_sc_hs/v0.0.1/models/udp_mux_2to1_n/sky130_fd_sc_hs__udp_mux_2to1_n.blackbox.v     | 6 +++---
 libraries/sky130_fd_sc_hs/v0.0.1/models/udp_mux_2to1_n/sky130_fd_sc_hs__udp_mux_2to1_n.v              | 6 +++---
 libraries/sky130_fd_sc_hs/v0.0.1/models/udp_mux_4to2/sky130_fd_sc_hs__udp_mux_4to2.blackbox.v         | 6 +++---
 libraries/sky130_fd_sc_hs/v0.0.1/models/udp_mux_4to2/sky130_fd_sc_hs__udp_mux_4to2.v                  | 6 +++---
 libraries/sky130_fd_sc_hvl/v0.0.0/models/udp_mux_4to2/sky130_fd_sc_hvl__udp_mux_4to2.blackbox.v       | 6 +++---
 libraries/sky130_fd_sc_hvl/v0.0.0/models/udp_mux_4to2/sky130_fd_sc_hvl__udp_mux_4to2.v                | 6 +++---
 libraries/sky130_fd_sc_hvl/v0.0.1/models/udp_mux_4to2/sky130_fd_sc_hvl__udp_mux_4to2.blackbox.v       | 6 +++---
 libraries/sky130_fd_sc_hvl/v0.0.1/models/udp_mux_4to2/sky130_fd_sc_hvl__udp_mux_4to2.v                | 6 +++---
 libraries/sky130_fd_sc_hvl/v0.0.2/models/udp_mux_4to2/sky130_fd_sc_hvl__udp_mux_4to2.blackbox.v       | 6 +++---
 libraries/sky130_fd_sc_hvl/v0.0.2/models/udp_mux_4to2/sky130_fd_sc_hvl__udp_mux_4to2.v                | 6 +++---
 libraries/sky130_fd_sc_lp/v0.0.0/models/udp_mux_2to1_n/sky130_fd_sc_lp__udp_mux_2to1_n.blackbox.v     | 6 +++---
 libraries/sky130_fd_sc_lp/v0.0.0/models/udp_mux_2to1_n/sky130_fd_sc_lp__udp_mux_2to1_n.v              | 6 +++---
 libraries/sky130_fd_sc_lp/v0.0.0/models/udp_mux_4to2/sky130_fd_sc_lp__udp_mux_4to2.blackbox.v         | 6 +++---
 libraries/sky130_fd_sc_lp/v0.0.0/models/udp_mux_4to2/sky130_fd_sc_lp__udp_mux_4to2.v                  | 6 +++---
 libraries/sky130_fd_sc_lp/v0.0.1/models/udp_mux_2to1_n/sky130_fd_sc_lp__udp_mux_2to1_n.blackbox.v     | 6 +++---
 libraries/sky130_fd_sc_lp/v0.0.1/models/udp_mux_2to1_n/sky130_fd_sc_lp__udp_mux_2to1_n.v              | 6 +++---
 libraries/sky130_fd_sc_lp/v0.0.1/models/udp_mux_4to2/sky130_fd_sc_lp__udp_mux_4to2.blackbox.v         | 6 +++---
 libraries/sky130_fd_sc_lp/v0.0.1/models/udp_mux_4to2/sky130_fd_sc_lp__udp_mux_4to2.v                  | 6 +++---
 libraries/sky130_fd_sc_ls/v0.1.0/models/udp_mux_2to1_n/sky130_fd_sc_ls__udp_mux_2to1_n.blackbox.v     | 6 +++---
 libraries/sky130_fd_sc_ls/v0.1.0/models/udp_mux_2to1_n/sky130_fd_sc_ls__udp_mux_2to1_n.v              | 6 +++---
 libraries/sky130_fd_sc_ls/v0.1.0/models/udp_mux_4to2/sky130_fd_sc_ls__udp_mux_4to2.blackbox.v         | 6 +++---
 libraries/sky130_fd_sc_ls/v0.1.0/models/udp_mux_4to2/sky130_fd_sc_ls__udp_mux_4to2.v                  | 6 +++---
 libraries/sky130_fd_sc_ls/v0.1.1/models/udp_mux_2to1_n/sky130_fd_sc_ls__udp_mux_2to1_n.blackbox.v     | 6 +++---
 libraries/sky130_fd_sc_ls/v0.1.1/models/udp_mux_2to1_n/sky130_fd_sc_ls__udp_mux_2to1_n.v              | 6 +++---
 libraries/sky130_fd_sc_ls/v0.1.1/models/udp_mux_4to2/sky130_fd_sc_ls__udp_mux_4to2.blackbox.v         | 6 +++---
 libraries/sky130_fd_sc_ls/v0.1.1/models/udp_mux_4to2/sky130_fd_sc_ls__udp_mux_4to2.v                  | 6 +++---
 libraries/sky130_fd_sc_ms/v0.0.0/models/udp_mux_2to1_n/sky130_fd_sc_ms__udp_mux_2to1_n.blackbox.v     | 6 +++---
[... cut ...]
 libraries/sky130_fd_sc_ms/v0.0.1/models/udp_mux_2to1_n/sky130_fd_sc_ms__udp_mux_2to1_n.blackbox.v     | 6 +++---
 libraries/sky130_fd_sc_ms/v0.0.1/models/udp_mux_2to1_n/sky130_fd_sc_ms__udp_mux_2to1_n.v              | 6 +++---
 libraries/sky130_fd_sc_ms/v0.0.1/models/udp_mux_4to2/sky130_fd_sc_ms__udp_mux_4to2.blackbox.v         | 6 +++---
 libraries/sky130_fd_sc_ms/v0.0.1/models/udp_mux_4to2/sky130_fd_sc_ms__udp_mux_4to2.v                  | 6 +++---
 54 files changed, 162 insertions(+), 162 deletions(-)
```

fangism, 2020-10-06:
Drive-by: reordering positional args like this would be a nice job for a proper refactoring tool.

mithro, 2020-10-06:
Since these files are auto-generated, a proper refactoring tool is probably not going to be all that helpful.

fangism, 2020-10-06:
> Since these files are auto-generated, a proper refactoring tool is probably not going to be all that helpful.

Well yes, fixing the generator when there is one is the right solution.

PR #167 (2020-10-06): verilog: Fix ordering of ports on mux primitives.
Verilog requires the first signal in a primitive's pin list must be the output.

Fixes #153.

 - Updating [`sky130_fd_sc_hd` latest](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/v0.0.2-38-g9b8d061b..v0.0.2-40-g836b7fa0) to v0.0.2-40-g836b7fa0
 - Updating [`sky130_fd_sc_hd` v0.0.1](https://foss-eda-tools.googlesource.com/skywater-pdk/libs/sky130_fd_sc_hd/+/v0.0.1-19-g23c9281bd..v0.0.1-20-g01fde9a43) to v0.0.1-20-g01fde9a43
 - Updating [`sky130_fd_sc_hd` v0.0.2](https://foss-eda-tools.googlesource.com/skywater-pdk/l [... cut]
closed 2020-10-06 by mithro