google/skywater-pdk #154: Verilog module sky130_fd_sc_hd__bleeder does not exist

RTimothyEdwards, 2020-09-25:
@mithro :  There exists a file cells/lpflow_bleeder/sky130_fd_sc__lpflow_bleeder.v but it does not contain any module definition.  But that module is called from inside sky130_fd_sc_hd__lpflow_bleeder_1.  The original source file, which is scs8hd_lpflow_bleeder_1.v, does have a definition (it contains a pulldown, which is pretty much its entire contents).  This definition is somehow getting lost.

RTimothyEdwards, 2020-10-06:
@mithro :  This was not corrected in the latest update?  This issue prevents the simulation of the Caravel harness chip. . .

RTimothyEdwards, 2020-11-09:
@mithro :  Here is a patch that fixes the issue:

[issue_154.zip](https://github.com/google/skywater-pdk/files/5513387/issue_154.zip)

mithro, 2020-11-10:
@RTimothyEdwards -- `issue_154.zip` contains the fix for #178 not the bleeder cell.

mithro, 2020-11-10:
Looks like the `issue_178.zip` contains the patch for this issue.

closed 2020-11-10 by mithro