google/skywater-pdk #158: Cleanup the symbols to have new names

mithro, 2020-09-30:
The pull request at https://github.com/google/skywater-pdk/pull/157 added the primitive models. A lot of the symbols images that were added still use the old non-SKY130 names. Specific examples are the diodes. These should be cleaned up and made to match the new names.

diadatp, 2020-11-12:
The symbol SVGs look like wrappers around PNGs with a few references to Inkscape(?), some invisible bullet point shape paths and glyphs for a font. I'm assuming the corrected symbols will be a complete replacement.

@StefanSchippers has created most of the required symbols and shared them [here](https://github.com/StefanSchippers/xschem_sky130/tree/main/sky130_fd_pr). Xschem has the ability to perform SVG exports. I've attached PNG versions as a preview. They differ from the SVG versions in the way the text is rendered. If the style is fine then the work of completing the set and generating them to the correct size can go forward.

[image]
[image]

StefanSchippers, 2020-11-13:
Device names should be definitely cleaned up in the 'Device Details' web page.
We have too generic cell names like 
'diode' 
to overly complex cell names like
'sky130_fd_pr__cap_vpp_XXpXxYYpY_{MM}(_shield(SS)*)(_float(FF)*)'
I have no problem changing the names. Information about Diodes in particular should be cleaned up.