google/skywater-pdk #169: ngspice does not apply tc1, tc2 to resistors with equation for r

RTimothyEdwards, 2020-10-06:
This is an ngspice error, maybe, or maybe just an inconsistency between ngspice and spectre.  But it has been discovered that resistors which have an equation in the model for the value of "r" will not get the temperature coefficient parameters "tc1" and "tc2" applied, so that in ngspice simulation, the resistance is completely flat over temperature, which is thoroughly unrealistic (or, to be blunt, just plain wrong).

Options:  (1) Check with Holger Vogt if this is intentional behavior on ngspice's part.  If this is determined to be an ngspice error, then it should be fixed in ngspice.  (2) If this is determined to be a "known inconsistency", then every "r=" equation for a modeled resistor needs to expand the equation to incorporate the tc1, tc2, and tnom values.  This is a known simple equation which is:

    R = R [1 + TC1 * (Temp - Tnom) + TC2 * (Temp - Tnom)^2]

dwarning, 2020-10-17:
Hi Tim,
I did not full understood: ngspice has tc1 and tc2 as instance parameter with the formular you have shown. Excerpt from manual:
RE1 1 2 800 newres dtemp=5
.MODEL newres R tc1=0.001

RE2 a b 1.4k tc1=2m tc2=1.4u

RE3 n1 n2 1Meg tce=700m (this is an alternative exponential temp model)

The attached file show this capabilities.

[temper-res-1.txt](https://github.com/google/skywater-pdk/files/5395145/temper-res-1.txt)

RTimothyEdwards, 2020-10-17:
@dwarning :  The issue here is not that ngspice does not implement TC1 and TC2.  The issue is that when a resistor has a model but ALSO defines an equation for R, then TC1 and TC2 are completely ignored.  To see that in your testbench, I think you would need to do "RE1 1 2 r=800 newres dtemp=5", although it might be required to have "r=" as an actual expression, such as ".param rbase=800" and "RE1 1 2 r='rbase*1' newres dtemp=5".

I have not investigated this problem myself, but my understanding from others who reported the problem is that the use of the explicit equation for "r=" in the resistor instance causes TC1 and TC2 to be ignored.

The real question for the ngspice developers is whether it is intended behavior that setting "R='(equation)'" in the component instance means that the equation describes the complete and entire behavior of the resistor, or if that equation should be subsequently inserted into the equation for temperature.  Apparently, the behavior of other simulators is to apply the temperature equation after evaluating any equation for R in the instance.  This seems like a reasonable behavior to me.  The values of TC1 and TC2 are part of the model and the equation for temperature variation is simply a polynomial curve fit;  if anyone really wanted to capture the temperature behavior of the device in the passed equation with some alternative expression, they could always set the model TC1 and TC2 to zero.

dwarning, 2020-10-17:
Thanks Tim,

now I understood. You want use equation + model. But this is not 
possible at the moment.

And it is to consider that we have a bsource conversion so far network 
values come into play.

Please see my attached example file and observe the output of the 
listing command. Perhaps one of the equation example fullfil your 
expectation.

r into bsource conversion

v1  33 0  0.0
v2 2 0 1.0

.param rbase = 800
.param ns = 10

*this works value + model
*R1  33 0 800 newres
.model newres R tc1=1e-3 tc2=1e-5

*not allowed equation + model
*R1  33 0 r='rbase*min(ns,5)' newres
.model newres R tc1=1e-3 tc2=1e-5

*this will work - no bsource conversion
*R1  33 0 'rbase*min(ns,5)' tc1=1e-3 tc2=1e-5

*this will work - with bsource conversion, tc's are considered
R1  33 0 'v(2)*rbase*min(ns,5)' tc1=1e-3 tc2=1e-5

*this will not work - bsource conversion, but model is ignored, uncomplaining
*R1  33 0 'v(2)*rbase*min(ns,5)' newres

.control
dc v1 0.1 10 0.1 temp 0 100 10
plot v(33)/-i(v1)
listing
.endc
.end

holvo, 2020-11-27:
Has been fixed in ngspice (git pre-master branch)

~~~
.param MM=3
.param PP=0
.param TC=-1.47e-3
.param sum={TC-MM*PP}

These are o.k.:
r1 1 0 r={1k*(1+v(1))} tc1={sum}
r2 2 0 r={1k*(1+v(2))} tc1 = {TC}
r3 3 0 r={1k*(1+v(2))} tc1 = -1.47e-3

this is not o.k. (no equation evaluation available in TC1={...})
r4 4 0 r={1k*(1+v(1))} tc1={TC-MM*PP}
~~~